morefigs/pymba #13: Liveview at 30 fps

Anki11, 2015-02-16:
Hi,

this wrapper is genius!
Thanks for sharing it to the world.

However, trying several things (and your live preview example), I can only loop with 30 fps, although my AVT Pike can stream with 200 fps. Accessing it via pymba and printing the AcquisitionFrameRate, it displays me something between 205 and 211 (what would be fine). And, when I trying to stream via the Vimba Viewer, I get my 200 fps.

Regardless of displaying it via opencv or not, it still remains at roughly 30 fps.
Is there a way to enhance this framerate?

Thanks for replying
Anki

morefigs, 2015-02-16:
Thanks, glad you like it :)

Looking at the examples in the tests folder, they all use the lines

    c0.runFeatureCommand("AcquisitionStart")
    c0.runFeatureCommand("AcquisitionStop")

on _each_ frame. I suspect this could be wrong and they should be used only once each (before and after acquiring _all_ of the frames). When you installed Vimba you would also have installed a few programming documentation PDFs from AVT that should hint at usage of commands (although not in Python of course). I'd try it myself but don't have an AVT camera any more.

The other thing to check is that your camera shutter/exposure is not too long as that would of course limit frame rate.

morefigs, 2015-02-16:
Just realised some examples also use the line

    camera0.AcquisitionMode = 'SingleFrame'

this should be `'Continuous'` instead for multiple images I believe, but again double check the AVT docs.

morefigs, 2015-02-16:
@fmder this issue might interest you as well.

derricw, 2015-02-16:
Yes, morefigs is absolutely right about the 'AquisitionStart' / 'AquisitionStop' commands.  Those should not be running every frame.

Moving them outside of the loop should increase your frame rate a bit, but if you are wanting to display as fast as possible, there will be more steps involved.  Specifically, you will have to queue the next frame with frame.queueFrameCapture() immediately after you run frame.waitFrameCapture().  Here is what the loop should look like to maximize FPS:

    c0.runFeatureCommand("AcquisitionStart")

    frame.queueFrameCapture()  #queue the first frame

    while 1:
        frame.waitFrameCapture(1000)
        frame.queueFrameCapture()
        frame_data = frame.getBufferByteData()
        img = np.ndarray(buffer=frame_data,
                         dtype=np.uint8,
                         shape=(frame.height,frame.width,1))
        cv2.imshow("test",img)
        framecount+=1
        k = cv2.waitKey(1)
        if k == 0x1b:
            cv2.destroyAllWindows()
            print "Frames displayed: %i"%framecount
            print "Frames dropped: %s"%droppedframes
            break

    c0.runFeatureCommand("AcquisitionStop")

You still will not achieve your maximum frame rate because you are displaying the frame in OpenCV in the same thread, and that takes some time, but if you were to comment out the cv2.imshow() line, you should be able to acquire at whatever the max frame rate of your camera is.

Anki11, 2015-02-17:
Hey, thanks for your fast reply!

Thank you for all the tips, I moved the acquisitationstart and stop outside the loop, my code looks basically like the one from derricw. The first thing I did was changing "Singleframe" to "Continuous".

Another I started to like is only showing every nth frame via openCV. This enhances the sampling rate and at the same time, I can follow what happens:

```python
if framecount % 7 == 0:
    cv2.imshow('live',frame)
    key = cv2.waitKey(1)
```

Using this and your tips, I get a framerate of ca. 170 fps. Probably I can enhance this even more, but at the moment I'm fine with this.

jferch, 2015-02-18:
Do you guys know a performant way to save the images acquired during the loop? If I run cv2.imwrite('foo{}.jpg'.format(framecount), img) my fps drops to around 10. Is there any other command I could use? 20ish fps would suffice. Thanks in advance...

Anki11, 2015-02-18:
Two things you might can do:
a) save the images in a multidimensional array and save them after the loop
b) grab a video using cv2 without any compression and compress it afterwards. Depending on the compression your framerate will drop again. Actually, I tried this yesterday. You may test some codecs, the Intel ones on my computer are capable of 60fps.

All the best

derricw, 2015-02-18:
Here are your options:

1) Write them to an HDF5 Dataset.
2) Write them to an numpy.ndarray in memory, then write them to files after acquisition is over.
3) Write them to tiff stacks, using [tifffile.py's](http://www.lfd.uci.edu/~gohlke/code/tifffile.py) `imsave()` function.

morefigs, 2015-02-19:
Closing as FPS issue solved.

closed 2015-02-19 by morefigs

Anki11, 2015-09-02:
Just a quick comment to jferch's post and derricw's answer 2). Numpy is terrible slow in saving data on the go, just use the in-built list function and append the frames. The memory allocation is fine for doing this with > 400 fps.