morefigs/pymba #14: simultaneous capturing with 2 cameras

jferch, 2015-03-14:
Hey guys, 

I'm trying to acquire simultaneous footage with two mantas using the precision time protocol supported by the cameras. I already got them to sync properly but am currently stuck getting the image data. Do you know a good way to handle the acquisition of two cameras? Would threading / dividing into two separate processes be a good idea? Any help is highly appreciated :). Thanks in advance.

Cheers

morefigs, 2015-03-17:
Can't say I've tried that before, and I'm not aware of the precision time protocol, but in general threading is a good idea for asynchronous events. If your threads are CPU intensive then move onto multiple processes (assuming you have multiple cores).

derricw, 2015-03-17:
I have done acquisition from two cameras and, like morefigs recommended, I found it easiest to just use two processes.  You can use the `subprocess` module for this.

jferch, 2015-03-23:
Thanks for the replies. Would you mind sharing your code? That would save me some time :>

jferch, 2015-03-24:
Nevermind I got it working!

closed 2015-03-24 by jferch