morefigs/pymba #15: Run on 32 bit or arm?

DanielArnett, 2015-07-08:
I am trying to run pymba on an arm based processor (Jetson Development board running the Tegra K1 Processor).  

Error Generated
`from vimbadll import VimbaDLL
  File "/usr/local/lib/python2.7/dist-packages/pymba-0.1-py2.7.egg/pymba/vimbadll.py", line 21, in <module>
    "GENICAM_GENTL64_PATH"), "you need your GENICAM_GENTL64_PATH environment set.  Make sure you have Vimba installed, and you have loaded the /etc/profile.d/ scripts"`

I tried editing `/usr/local/lib/python2.7/dist-packages/pymba-0.1-py2.7.egg/pymba/vimbadll.py` and changing

`else:
    _cruntime = CDLL("libc.so.6")
    dll_loader = cdll
    assert os.environ.get(
        "GENICAM_GENTL64_PATH"), "you need your GENICAM_GENTL64_PATH environment set.  Make sure you have Vimba installed, and you have loaded the /etc/profile.d/ scripts"
    vimba_dir = "/".join(os.environ.get("GENICAM_GENTL64_PATH").split("/")
                         [1:-3])
    vimbaC_path = "/" + vimba_dir + "/VimbaC/DynamicLib/x86_64bit/libVimbaC.so"`

to 

`else:
    _cruntime = CDLL("libc.so.6")
    dll_loader = cdll
    assert os.environ.get(
        "GENICAM_GENTL32_PATH"), "you need your GENICAM_GENTL32_PATH environment set.  Make sure you have Vimba installed, and you have loaded the /etc/profile.d/ scripts"
    vimba_dir = "/".join(os.environ.get("GENICAM_GENTL32_PATH").split("/")
                         [1:-3])
    vimbaC_path = "/" + vimba_dir + "/VimbaC/DynamicLib/arm_32bit/libVimbaC.so"`

I thought this would simply change the .so file used from the 64 bit version to the 32 bit version but that only generated a new error:

`Traceback (most recent call last):
  File "vimba-streamer.py", line 1, in <module>
    from pymba import *
  File "/usr/local/lib/python2.7/dist-packages/pymba-0.1-py2.7.egg/pymba/__init__.py", line 2, in <module>
    from vimba import Vimba
  File "/usr/local/lib/python2.7/dist-packages/pymba-0.1-py2.7.egg/pymba/vimba.py", line 3, in <module>
    from vimbadll import VimbaDLL
  File "/usr/local/lib/python2.7/dist-packages/pymba-0.1-py2.7.egg/pymba/vimbadll.py", line 30, in <module>
    class VimbaDLL(object):
  File "/usr/local/lib/python2.7/dist-packages/pymba-0.1-py2.7.egg/pymba/vimbadll.py", line 112, in VimbaDLL
    _vimbaDLL = dll_loader.LoadLibrary(vimbaC_path)
  File "/usr/lib/python2.7/ctypes/__init__.py", line 443, in LoadLibrary
    return self._dlltype(name)
  File "/usr/lib/python2.7/ctypes/__init__.py", line 365, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /opt/Vimba_1_3/VimbaC/DynamicLib/arm_32bit/libVimbaC.so: cannot open shared object file: No such file or directory`

I checked and yes, there is a such file or directory as `/opt/Vimba_1_3/VimbaC/DynamicLib/arm_32bit/libVimbaC.so`, but regardless this does not work. Does anyone have any thoughts or suggestions? Thank you.

rbogle, 2015-07-08:
I posted a pull request from my fork. It includes an added code block to do exactly what you tried with the 32-bit lib. It works for the x86_32. Perhaps there's a permission problem on the directory in question?

DanielArnett, 2015-07-08:
Thanks for the quick response. I tried out the changes you made. The only problem is that I'm running on arm build so I had to change the last line 

`if '64' in platform.architecture()[0]:
        assert os.environ.get(
            "GENICAM_GENTL64_PATH"), "you need your GENICAM_GENTL64_PATH environment set.  Make sure you have Vimba installed, and you have loaded the /etc/profile.d/ scripts"
        vimba_dir = "/".join(os.environ.get("GENICAM_GENTL64_PATH").split("/")
                             [1:-3])
        vimbaC_path = "/" + vimba_dir + "/VimbaC/DynamicLib/x86_64bit/libVimbaC.so"
    else:
        assert os.environ.get(
            "GENICAM_GENTL32_PATH"), "you need your GENICAM_GENTL32_PATH environment set.  Make sure you have Vimba installed, and you have loaded the /etc/profile.d/ scripts"
        vimba_dir = "/".join(os.environ.get("GENICAM_GENTL32_PATH").split("/")
                             [1:-3])
        vimbaC_path = "/" + vimba_dir + "/VimbaC/DynamicLib/arm_32bit/libVimbaC.so"`

I've set the .so to be `rwx`, and it's definitely in the location that vimbaC_path is checking, but I'm still getting the `No such file or directory` error.

DanielArnett, 2015-07-12:
I found the problem.  For arm processors [Vimba](http://www.alliedvision.com/en/products/software.html) has a "Hard Float" and a "Soft Float" library. The arm processor that I am using (Tegra K1 on a Jetson Development board) has a Hardware floating point unit. 

(Note: if you want to check your arm system try running `cat /proc/cpuinfo | grep vfp` in the terminal. If you see the `vfp` flag anywhere in /proc/cpuinfo, then you have a hardware floating point unit (FPU).)

The short of it is that if you're running an arm processor and getting an `OSError:`...`: cannot open shared object file: No such file or directory` error, chances are you might have a hardware FPU and are calling a library compiled for floating point operations in software. 

```python
>>> from ctypes import *
>>> CDLL("/lib/arm-linux-gnueabi/ld-2.19.so")
```
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/ctypes/\__init\__.py", line 365, in \__init\__
    self._handle = _dlopen(self._name, mode)
OSError: /lib/arm-linux-gnueabi/ld-2.19.so: cannot open shared object file: No such file or directory
```
```python
>>> CDLL("/lib/arm-linux-gnueabihf/ld-2.19.so")
<CDLL '/lib/arm-linux-gnueabihf/ld-2.19.so', handle 31a688 at b6c7da50>
>>> 
```

In that exmaple I tried importing a software-float library. Notice the terrible error `OSError`...`.so: cannot open shared object file: No such file or directory`. I've found that this awful error is thrown every time python tries to import .so files that use software floats instead of hardware floats. Using the [`readelf` command](https://blogs.oracle.com/jtc/entry/is_it_armhf_or_armel) I can check to see if the library uses [`VFP`, arm's vector floating-point hardware](http://www.arm.com/products/processors/technologies/vector-floating-point.php). 
```
ubuntu@tegra-ubuntu:~/andy/vimba-mjpeg$ readelf -A /lib/arm-linux-gnueabi/ld-2.19.so | grep Tag_ABI_VFP_args
ubuntu@tegra-ubuntu:~/andy/vimba-mjpeg$ readelf -A /lib/arm-linux-gnueabihf/ld-2.19.so | grep Tag_ABI_VFP_args
  Tag_ABI_VFP_args: VFP registers
```
Here you can see that the shared object files in `/lib/arm-linux/gnueabi/` directory don't use VFP, whereas the files in `/lib/arm-linux-gnueabihf/` directory do. 

I log all this here in case someone else runs into this problem, but I would like to know if there's a better place for me to log it, or even if I could submit a bug report to the Python Software Foundation.

closed 2015-07-12 by DanielArnett