morefigs/pymba #21: calling getFeatureNames() raises an exception with the new drivers

nikitchm, 2016-02-01:
Hello,

After the installation of Vimba v1.4 from v1.3 and re-installation of pymba from this git repository, I started getting an error to any call to a function using getFeatureNames():
import pymba
vimba = pymba.Vimba()
vimba.startup()
system = vimba.getSystem()
system.getFeatureNames()
ERROR:
  File "...\Anaconda\lib\site-packages\pymba-0.1-py2.7.egg\pymba\vimbaobject.py", line 109, in getFeatureNames
    return list(featInfo.name for featInfo in self._getFeatureInfos())
  File "...\Anaconda\lib\site-packages\pymba-0.1-py2.7.egg\pymba\vimbaobject.py", line 81, in _getFeatureInfos
    raise VimbaException(errorCode)
VimbaException: More data was returned in a string/list than space was provided.

Would anyone know how to fix it?
Thank you,
Max

morefigs, 2016-02-01:
This is the code (in `vimbaobject.py`) that's throwing the error:

    # call once to get number of available features
    # Vimba DLL will return an error code
    errorCode = VimbaDLL.featuresList(self._handle,
                                      byref(dummyFeatureInfo),
                                      0,
                                      byref(numFound),
                                      sizeof(dummyFeatureInfo))
    if errorCode != 0:
        raise VimbaException(errorCode)

As a quick guess, you could try changing the `0` to something else. The error is complaining about list size, and that integer is the `listLength` argument. Perhaps that parameter has become stricter in the latest version of Vimba.

You could also try just ignoring the error altogether.

Sorry for the sloppy answer, but I don't have a camera or even a Windows environment to test on at the moment. Let me know if you can't get it working and I'll have a closer look.

nikitchm, 2016-02-02:
Thank you very much for the prompt response!

Ignoring the error still resulted in an error. DLL returns numFeatures = 4294967295, which is obviously wrong. Setting it to 1000 (finger into the sky), seems to have solved the problem!

The strange thing is that the Vimba viewer works fine. Do you think there might be a better fix to the problem?

Thank you again!

morefigs, 2016-02-02:
This issue is caused by how I chose parameter values for that function, so the issue is only related to pymba. 

Glad that fix worked for you!

jrast, 2016-03-01:
The VimbaC docs mention how the function should be used:

> This method lists all implemented features, whether they are currently available
> or not. The list of features does not change as long as the camera/interface
> is connected. "pNumFound" returns the number of VmbFeatureInfo elements.
> This function is usually called twice: once with an empty list to query the length
> of the list, and then again with an list of the correct length.

So the function must be called twice.

morefigs, 2016-03-06:
Changed the value for the first call of listLength from 0 to 999.

closed 2016-03-06 by morefigs