morefigs/pymba #24: error running system.GeVTLIsPresent

dbstrasf, 2016-03-05:
I am getting the following error when trying the following code. I am running python 2.7.11 and Vimba 1.4. In vimbaobject.py. I changed the VimbaDLL.featuresList parameter from 0 to 1000. Any thoughts?

from pymba import *
import time

with Vimba() as vimba:
    # get system object
    system = vimba.getSystem()

if system.GeVTLIsPresent:
        system.runFeatureCommand("GeVDiscoveryAllOnce")
        time.sleep(0.2)

This is the error:

Traceback (most recent call last):
  File "<pyshell#6>", line 1, in <module>
    if system.GeVTLIsPresent:
  File "C:\Python27\lib\pymba\vimbaobject.py", line 35, in __getattr__
    if attr in self.getFeatureNames():
  File "C:\Python27\lib\pymba\vimbaobject.py", line 109, in getFeatureNames
    return list(featInfo.name for featInfo in self._getFeatureInfos())
  File "C:\Python27\lib\pymba\vimbaobject.py", line 81, in _getFeatureInfos
    raise VimbaException(errorCode)
VimbaException: VmbStartup() was not called before the current command.

jferch, 2016-03-05:
Well, as it says in the last line, startup was not called. Try something like this:

vimba = Vimba()
vimba.startup()
system = vimba.getSystem()
system.runFeatureCommand("GeVDiscoveryAllOnce")
time.sleep(0.2)

Cheers

dbstrasf, 2016-03-05:
Thanks for the quick reply. That looks to be working. Now my issue is that
when I try to get feature names after I open the camera, Idle crashes. Any
thoughts on that? The following is what I'm running:

from pymba import *
import time

vimba = Vimba()
vimba.startup()
system = vimba.getSystem()
system.runFeatureCommand("GeVDiscoveryAllOnce")
time.sleep(0.2)

cameraIds = vimba.getCameraIds()
for cameraId in cameraIds:
    print 'Camera ID:', cameraId

# get and open a camera
#camera0 = vimba.getCamera(cameraIds[0])
#camera0.openCamera()

# list camera features
cameraFeatureNames = camera0.getFeatureNames()

morefigs, 2016-03-06:
`getFeatureNames` is broken at the moment, see https://github.com/morefigs/pymba/issues/21

morefigs, 2016-03-25:
This issue should now be resolved, see https://github.com/morefigs/pymba/pull/27

closed 2016-03-25 by morefigs