morefigs/pymba #25: runFeatureCommand requires bytestrings

Gloweye, 2016-03-10:
Note: I am using Vimba 1.4. It is possible that syntax was changed here.

On all examples found here, normal strings seem to be supplied. However, finally succeeding a call to `system.getFeatureNames()` gave me all responses in bytestrings. Why this difference exists I do not know, but it got a lot of stuff running. 

An example is `system.GeVTLIsPresent`, which seems to require a bytestring as well. Replacing it with an explicit `system.__getattr__(b"GeVTLIsPresent")` seems to get it working without a hitch.

Since I didn't find any mention, I decided to make it a new issue. I don't know if making internal changes will be compatible with older versions of vimba, but wanted the knowledge out here.

jrast, 2016-05-04:
Which Version of Python are you using?

I use things like ``system.GeVTLIsPresent`` all over my code without any problems.

Gloweye, 2016-05-04:
Currently switched to Vimba 2.0, using Python 3.5. Was using 3.4 back then.

Issue is gone, but that could also be because I'm using the python 3 branch.

Anyway, it's working right now without the explicit calls like I mentioned up there.

closed 2017-05-05 by morefigs