morefigs/pymba #28: Black Lines on readings

Gloweye, 2016-03-31:
I've been getting black lines on my pictures lately, and I haven't been able to find out why. 

Script:
frame is...a frame. image is externally triggered, but I've also seen it happen in internal triggers. All of this is done in it's own dedicated thread. 
```
#Here's a lot of setup code
a = frame.waitFrameCapture(timeout = self.timeout*1000)# convert to milliseconds
print("Errorcode:",a)
timer_abort_queue.put((timerID,)) #Logs the duration taken from the main thread
cameraref.endCapture()
try:
     b = np.ndarray(buffer = frame.getBufferByteData(),
                              dtype = np.uint8,
                              shape = (frame.width,frame.height))
     print(b.shape)
     print(b)
 except:
      raise
cameraref.revokeAllFrames()
if a == 0:
      self.queue.put((identifierstring,copy.deepcopy(b)))  #gives the array to the main thread so enable processing of the data
```

Example of a black lined result is like this:
[image]

What I considered:
  Data corruption due to some pointer getting into a fight with python data management: Seems unlikely to me. However, it would explain why it's a solid line across multiple rows, stopping at apparently random intervals of rows. 
  Camera not reading out entire sensor: While the camera is capable, I haven't told it to, and it does seem to be able to only select rectangles, not these semi-random lines seen in the image.
  Camera not sending entire sensor readout over the GigE: Appears weird to me. I'd have expected to see vimbaDLL complain about it, or the data to be misarranged instead of partially missing. Also, `StatFrameDropped`, `StatPacketMissed`, `StatPacketResent` and `StatFrameRescued` all return 0 after taking the above image.
  Data corruption after taking the image(somewhere in the numpy array): it's printed right after and shows the values belonging to the lines there. Also, deepcopy should prevent any loss due to referencing issues.

Side note: Vimba Viewer itself doesn't seem to exhibit this problem. Or maybe it ignores to faulty frames someway.

I couldn't find any issues like this anywhere. If there's someone who has an idea, I'd love to hear it.

morefigs, 2016-03-31:
Do you get the same problem if you run a super simple script (like that in the Pymba readme) without threading, etc? Your camera calls are slightly different too, e.g. `endCapture()` is called before you access the frame (I've no idea how much that matters however).

rbogle, 2016-03-31:
I found this problem on my jai cameras using the GigE via python-aravis. Ultimately it was network latency causing packet timeouts and incomplete frame buffers. I dont remember with pymba, can you check the buffer status to see if the buffer is complete/ready? Non-specific to pymba: I also was able to increase the packet delay & packet size (GevSCPD, GevSCPSPacketSize) in the GigE protocol to make packet timeouts much less likely. Lastly check the MTU of your eth adapter and see if you can bump that up, perhaps it got reset?

morefigs, 2016-03-31:
Also ensure you are using 1000Mbps (not 10/100Mbps) switches and network cards.

Gloweye, 2016-04-06:
I seem to be unable to replicate it right now, using the exact same script and settings. If I find a more reliable way, I'll let all of you know. For the rest, i just suggest we keep our eyes open if something like it happens.

closed 2016-04-06 by Gloweye