morefigs/pymba #29: python 3.5

hstarmans, 2016-04-06:
Dear all,

I am using a modified version of Pymba for Python 3.5.
There are several challenges if Pymba is ported to Python 3.5.1, Windows 7, 64 bit
**major issue:**
```python
 from ctypes.util import find_msvcrt
 find_mscvct()
```
This code, located in vimbadll, does not work in Python 3.5.1
Python 3.5 is build against 'msvcr100.dll', for some reason, the find_mscvcrt does not work and returns "none". I know fixed it with:
```python
_cruntime = cdll.LoadLibrary('msvcr100.dll')
```
**minor issues:**
```python
print "something" should be print("something")
```
byte strings --> in python 3 you need to use encode and decode.

Best,

Rik

PS.
Due to the setup i have to use Windows 7 64 bit and can't use something like linux :-(.

mabl, 2016-04-20:
Hi @hstarmans, please have a look at #22 & #19, which address these problems. Feel free to test the branch and correct the reaming errors there.

draustin, 2016-06-20:
Hi all,

Thank you @mabl for providing this.

I have tested #22 on Python 3.5.1, Win 64, Vimba 1.4, Guppy PRO F201B. All tests and demos worked except test_interfaces, which crashes on the final line interface0.closeInterface().

Since I'm new to github I'll be specific about what I did: I ran
git fetch origin pull/22/head:python3
and reinstalled pymba.

hstarmans, 2016-06-22:
Dear all,
My apologies for my late response. Thank you @mabl, your solution was better than the temporary fix I used.
@draustin , thank you for your help with Git. I have tested it with Python 3.5.1 and win32. I had to resolve one merge conflict in pympa.py.
I still need to run the test and will update this post with test result once I have done this.

hstarmans, 2016-10-26:
Dear all,
I have tried the code from @mabl. The code partly works on Windows 7 64 bit with vimba 2.0. 
I have to start the vimba viewer, connect to the camera, click play, click stop and close the viewer.
If I do not execute this procedure the code does not work and I get a timeout, i.e. error code -12.
I use an ethernet camera GB2450 gigE.
```python
from pymba import Vimba
from time import sleep
vimba=Vimba()
vimba.startup()
vimsys=vimba.getSystem()
if vimsys.GeVTLIsPresent:
    vimsys.runFeatureCommand("GeVDiscoveryAllOnce")
    sleep(0.2)
camera0=vimba.getCamera(vimba.getCameraIds()[0])
camera0.openCamera()
camera0.AcquisitionMode='SingleFrame'
camera0.ExposureTimeAbs=6000  # duration in microseconds
# create new frames for the camera
frame0 = camera0.getFrame()    
# announce frame
frame0.announceFrame()
# capture a camera image
camera0.startCapture()
frame0.queueFrameCapture()
camera0.runFeatureCommand('AcquisitionStart')
sleep(0.2) # was suggested in other comment, but don't think it is needed
camera0.runFeatureCommand('AcquisitionStop')
result=frame0.waitFrameCapture()
if result!=0:
    print("First; open vimba viewer,connect to camera, click play, click stop and closer viewer otherwise this code does not work")
```

mabl, 2016-10-26:
Does this code work for you with stock master branch? I find it difficult to believe that device discovery would differ with the python 3 branch. However, I do not have a network camera, so no idea if your code is fully correct...

hstarmans, 2016-10-27:
@mabl, you are correct; the code also does not work with the stock master branch. The problem does not seem to be related to the modifications in the Python 3 branch. I have run the tests provided with xstools with both Python 2 and python 3. The opencv-example.py only works if I have started and closed the vimba viewer before executing the test.

morefigs, 2017-03-04:
Try again, code has changed slightly.

hstarmans, 2017-03-04:
Thanks for the changes. I do, however, no longer use the camera. I was using the code for a camera in a  3D printer from my previous employer.  I have mailed a former colleague; but can't test it myself.