morefigs/pymba #39: hardware trigger

ypark137, 2016-05-25:
Hi, I have been having problems with Line 2 trigger.  Line 2 trigger is working well on Vimba Viewer 2.0. The code was working before but it stopped working. I've tried restarting the camera and reinstalling pymba and vimba..My camera is GS1380. 

from pymba import *
import cv2
import numpy as np

vimba = Vimba()

vimba.startup()

system = vimba.getSystem()
system.runFeatureCommand("GeVDiscoveryAllOnce")

cameraIds = vimba.getCameraIds()

camera = vimba.getCamera(cameraIds[0])

camera.openCamera()

camera.TriggerMode = 'On'
camera.TriggerSource = 'Line2'
camera.TriggerActivation = 'RisingEdge' 
camera.PixelFormat = 'Mono8' 
camera.TriggerSelector='FrameStart'
camera.AcquisitionMode ='Continuous' 

print 'TriggerMode:', camera.TriggerMode
print 'TriggerSource:', camera.TriggerSource
print 'TriggerActivation:', camera.TriggerActivation
print 'TriggerSelector:', camera.TriggerSelector

frame = camera.getFrame()

frame.announceFrame()

camera.startCapture()

frame.queueFrameCapture()

frame.waitFrameCapture(3000)

print 'frame 1 captured or timed out '

imgData1 = frame.getBufferByteData()

camera.endCapture()

data1_np = np.ndarray(buffer=imgData1,
                           dtype=np.uint8,
                           shape=(frame.height, frame.width))

rgb = cv2.cvtColor(data1_np, cv2.COLOR_BAYER_RG2RGB)

cv2.imwrite( 'testTrigger.png'.format(1), rgb)
camera.revokeAllFrames()
camera.closeCamera()
print 'frame is saved as png'

morefigs, 2016-05-26:
Does adding `camera.runFeatureCommand('AcquisitionStart')` right after `frame.queueFrameCapture()` and `camera.runFeatureCommand('AcquisitionStop')` after `frame.waitFrameCapture(3000)` help?

ypark137, 2016-05-26:
It works! Thanks!

morefigs, 2016-05-26:
Glad to hear it =)

closed 2016-05-26 by morefigs

lsgcv, 2020-02-10:
Hi, 
since this was opened in 2016, can you provide an updated example?
I noticed that the method names listed above have changed since 2016.

Thank you!