morefigs/pymba #40: failing to import pymba

mooneyd22, 2016-10-12:
Hi, 

I encounter an error while running this very short code:

_#!/usr/bin/env python

import pymba
from pymba import *

with Vimba() as vimba:
    print vimba.getVersion()_

**File "test.py", line 3, in <module>
    import pymba
  File "/home/dylan/Python/pymba/pymba/__init__.py", line 2, in <module>
    from vimba import Vimba
  File "/home/dylan/Python/pymba/pymba/vimba.py", line 3, in <module>
    from vimbadll import VimbaDLL
  File "/home/dylan/Python/pymba/pymba/vimbadll.py", line 44, in <module>
    with open(vimbaC_path) as thefile:
IOError: [Errno 2] No such file or directory: '/home/dylan/Vimba_2_0/VimbaGigETL/CTI/x86_64bit:/home/dylan/Vimba_2_0/VimbaC/DynamicLib/x86_64bit/libVimbaC.so'**

When I first installed pymba, this code worked and output was the version number of Vimba, 1.5.0.
However, it now gives me the error above. I have installed the GigE and USB transport layers, and the path leading to the file is correct. I cannot fix this problem, I cannot think of any changes I made. The code was working and now it is not.

I have tried  deleting pymba, uninstalling the transport layers, and deleting Vimba, then re-installing everything again, but no luck.

I'm using Linux Mint Debian Edition and Python version 2.7. Processor is 1.7 GHz.

morefigs, 2016-10-12:
Can you manually check the path `'/home/dylan/Vimba_2_0/VimbaGigETL/CTI/x86_64bit:/home/dylan/Vimba_2_0/VimbaC/DynamicLib/x86_64bit/libVimbaC.so` exists?

morefigs, 2016-10-12:
That path obviously looks wrong, looks like you're getting two paths returned from the line

    vimba_dir = "/".join(os.environ.get("GENICAM_GENTL64_PATH").split("/")[1:-3])

in `vimbadll.py`. Not sure why. 

The quick fix would be to manually put your path into `vimbadll.py`. Can you post the output of `os.environ.get("GENICAM_GENTL64_PATH")` on your system?

mooneyd22, 2016-10-12:
I get `':/home/dylan/Vimba_2_0/VimbaGigETL/CTI/x86_64bit:/home/dylan/Vimba_2_0/VimbaUSBTL/CTI/x86_64bit'`

I found if uninstall the USB transport layer and only have GigETL installed, I can import pymba.
`GENICAM_GENTL64_PATH` is` :/home/dylan/Vimba_2_0/VimbaGigETL/CTI/x86_64bit` in this case.

seahawk67, 2017-09-05:
I know it has been a while ago. But if you are still interested in a solution please take a look at #58.