morefigs/pymba #46: Looking for some Pymba documentation

CrocodileDandy, 2017-04-27:
Hi,

First, thanks a lot for Pymba, it makes my job easier. We need more wrappers like this.
I have a some Goldeye P-008 and P-032 I try to automate and I find myself lacking documentation when using Pymba. I have the Vimba manual, a 2013 Vimba C API manual V1.2 along with its reference and the GigE features reference. Still I find myself struggling to understand what Pymba feature does what and how to call it.

For instance, when I do `vimba.getCameraIds()` I would love to know which ones are busy and can't be accessed. In the absence of documentation I find myself digging in the wrapper:

* There is a vimba feature for what I want: `vimba.getCameraInfo('device_ID')`. It picks an object in a list provided by `vimba._getCameraInfos()` when the ID matches the requested one. The permitted access mode can finally be retrieved by looking at the attribute of the info object: `info.permittedAccess`, which returns an long integer.

* Now I look into `vimbadll.py` to see that `VimbaDLL.camerasList(cameraInfoArray,numCameras,byref(numFound),sizeof(dummyCameraInfo))` that was called in `vimba._getCameraInfos()` to see that `VimbaDLL.camerasList()` is eventually defined as `_vimbaDLL.VmbCamerasList` which is a C function of the API.

* I open my Vimba C API function reference where the function definition lies but there is unfortunately nothing about the meaning of the long integer of `info.permittedAccess`.

Do you have any idea about where I can find this information ?

I feel like I am doing a lot of retro-engineering for a feature that seems pretty necessary to anyone using at least one shared camera. I feel the wrapper would really benefit of some extended documentation.
I would love to help on my spare time but I am a total beginner at wrapper development. I can read some C and I know a bit of Python though my object oriented game is far from state of the art.
How can I help ? Write docstrings ? Pull-modify-commit-push ?

Much <3

morefigs, 2017-05-05:
Hey, no worries, glad it's useful.

In a nutshell Pymba simply wraps the Vimba Windows DLL (or equivalent on other platforms) and makes a one to one mapping of the DLL's functions to Python functions. This is all done in vimbadll.py. There's really not much more to it than that. The other modules simply provide convenient classes that are useful for grouping together similar functions - all the camera functions are in the camera class for instance.

As such there's not really any documentation for Pymba apart from the Vimba API manual, etc you've already looked at. Looking around though it looks like this is what you need: http://docs.ros.org/hydro/api/avt_vimba_camera/html/VimbaC_8h_source.html#l00118. It describes the integer as type `VmbAccessModeType` and gives a description of the values.

Most docstrings are done, but yes any contributions would be more than welcome!

CrocodileDandy, 2017-05-11:
Hey,

Thanks so much for your answer.

I do get the wrapping nature of Pymba. I just tend to think that the C API manual and function reference aren't always the most accessible documentation. IMHO, a Python wrapper makes things so user-friendly that it would deserve a nice doc for itself with references to the C API doc for full details. Just food for thought.

Thanks a lot for the header you suggested. Did you just find it googling ?
Unfortunately, it seems different from my Vimba install. Indeed, when my camera can be accessed in full access mode, `vimba.getCameraInfo('device_ID').permittedAccess` returns `15L` instead of just `1L`.
Can I access my own `VimbaC.h` to check its content ? Should I decompile my own `VimbaC.dll` ?

Also, I use those cameras for generating liveviews. Is it possible to exploit the cameras in an asynchronous way with Pymba ? Or does Pymba only allows the frames to be fetched synchronously at this point ?
I can't give a definitive answer to this question while looking at `pymba.vimbaframe.py`.

With my best thoughts.

PS : I have written a little Python module defining a camera class working as a back end. It i meant to facilitate camera management in larger projects. Would it be appropriate to add commit/push that to the Pymba project or should I go host that elsewhere ?

seahawk67, 2017-05-12:
Hi,
I do not think that the `permittedAccess` will help you. This is static information about the modes a camera can be opened and is modeled as flags. So `15L` means you can in general use any mode to open the camera: full, read-only, etc. But, this does not give you any information on whether the camera is currently being controlled by another application. The only way to check this is to try to open the camera. If it fails with reason -6, the camera is in use.

CrocodileDandy, 2017-05-12:
@seahawk67 : Actually, it does. I use it to sort busy and available cameras on my network and offer the front end user to open only available cameras.

I haven't check the error code if I try to open a camera already opened in full access mode by another application. But my C Manual indeed lists an error `-6` in case > Operation is invalid with the current access mode

Much classes and <3

morefigs, 2017-05-29:
> a Python wrapper makes things so user-friendly that it would deserve a nice doc for itself with references to the C API doc for full details. 

Yes, agreed, it's just not something I have a lot of time for unfortunately.

> Thanks a lot for the header you suggested. Did you just find it googling ?

Yes.

> Unfortunately, it seems different from my Vimba install. Indeed, when my camera can be accessed in full access mode, vimba.getCameraInfo('device_ID').permittedAccess returns 15L instead of just 1L.
> Can I access my own VimbaC.h to check its content ? Should I decompile my own VimbaC.dll ?

I wouldn't bother trying to decompile things. As you say documentation is a little sparse so see what you can google or find in your Vimba install.

> Also, I use those cameras for generating liveviews. Is it possible to exploit the cameras in an asynchronous way with Pymba ? Or does Pymba only allows the frames to be fetched synchronously at this point ?

It's been a while, but yes it should be possible. You should be able to tell several cameras to take an image and then go and fetch each image afterwards, which would be sort of asynchronous. Beyond that just use threading.

> PS : I have written a little Python module defining a camera class working as a back end. It is meant to facilitate camera management in larger projects. Would it be appropriate to add commit/push that to the Pymba project or should I go host that elsewhere ?

That's probably best hosted in another repo. It could have Pymba as a dependency though.

robmarkcole, 2017-08-11:
@CrocodileDandy I would be interested to view your Camera class, is it online?
I aim to create helper functions to display the camera video feed within a jupyter notebook, and create a small GUI with buttons for capturing the image and performing operations such as subtracting a background

CrocodileDandy, 2017-08-13:
@robmarkcole Hey, unfortunately, not yet. I just started to re-write it from scratch last week because my code was too model-specific (and a bit shitty to be honest). Should be ready soon enough though.
I asked my management the permission to publish this. I will update this situation next week when we're all back from chill vacations.

**Also, regarding documentation in general**
One should always ask its camera vendor (AVT or someone else) for the full C API documentation containing the function reference. Pymba is then transparent enough to replicate the way cameras are handled in C.
@morefigs I happen to have the latest C API documentation. Maybe it could be helpful to add it to pymba's files ? Or at least host it somewhere and put a link to it in the README.TXT ?

morefigs, 2019-01-30:
I'll probably leave Vimba files / documentation out of Pymba for now, as they are easily installed with the Vimba installer. Let me know if this is still an issue.

closed 2019-01-30 by morefigs