morefigs/pymba #47: How to load camera config file

Gunboule, 2017-06-13:
Hi,
I just found your project and it works quite well. I'm just wondering if i can load a xml config file for a camera.
Sorry if it's a stupid question but i don't know much about this kind of programming.

Thanks !

CrocodileDandy, 2017-08-13:
Hey,

I doubt you can do this directly. The configuration of a camera can be set by changing the values of what are know as 'Features'. Those 'Features' are dependent upon each camera model. You can get the full list of them with the `getFeaturesNames()` method of the camera object of pymba. You can also access more information for each of the features with another method.

Pymba let you access those features as attributes of the camera object. So you can simply get the value of a feature with `camera.feature` and change a value with `camera.feature = value` assuming your value is valid for the feature.

So I'm afraid you need to [parse your xml file](https://stackoverflow.com/questions/1912434/how-do-i-parse-xml-in-python), maybe check the validity of the values and then address each feature individually. I feel like you're good to write your own parsing/addressing function/method.

Much <3

morefigs, 2017-08-13:
The Vimba C manual mentions:

> Additionally to the user sets stored inside the cameras, you can save the feature values as an XML file to
your host PC. For example, you can configure your camera with Vimba Viewer, save the settings as a file,
and load them with Vimba API. To do this, use the functions VmbCameraSettingsLoad and
VmbCameraSettingsSave.

But pymba does not yet implement these functions.

markjay4k, 2017-11-11:
I noticed that some Features can't be set using `camera.feature = value`.
For example, when I try to change the `GevCurrentIPAddress` value, I get the following error

```
VimbaException: Operation is invalid with the current access mode.
```

any suggestions on how to change the camera IP Address and Subnet Mask?

Thanks!

ouyangkid, 2018-04-22:
@markjay4k ip can be changed with the official tool vimba view in both windows and linux, I prefer you use windows first, when you get a camera first time. The ip maybe 169.*****
@Gunboule i list some of the parameters I used 
def ShowCurrentConfig(camera):
    cameraFeatureNames = camera.getFeatureNames()
    #print('here is all the configurations:')
    print(cameraFeatureNames)
    print("Details of what we need (maybe):")
    print('AcquisitionMode:', camera.AcquisitionMode)
    print('TriggerSource:', camera.TriggerSource)
    print('PixelFormat:', camera.PixelFormat)
    print('ExposureAuto', camera.ExposureAuto)
    print('ExposureAutoMin', camera.ExposureAutoMin)
    print('ExposureAutoMax', camera.ExposureAutoMax)
    print('GainAuto', camera.GainAuto)
    print('BalanceWhiteAuto',camera.BalanceWhiteAuto)
    print('OffsetX',camera.OffsetX)
    print('OffsetY', camera.OffsetY)
    print('Height', camera.Height)
    print('Width', camera.Width)

LIBINANDHINI, 2024-08-01:
I am using Allied vision camera for my project. But I don't know how to enable output configuration programmatically as I have my TriggerSource = Line3 and set Output LineSource=ine3Signal . I want to enable output line only when the output is passes and deactivates that when output is failed. Can anyone help me with this?