morefigs/pymba #48: Is there any way to handle events or external triggers ?

lonedevwolf, 2017-06-24:
i am not a very good programmer to start with, ive been going over and over the files and documentatiton, but i cannot find a way to recieve events or external triggers. can anybody help me ?

also if anyone is interested i have a mako g030b camera that i can provide access to for development purposes while im not using it (~12h per day)

morefigs, 2017-06-29:
None that I'm aware of.

Use of a camera when you're not using it would be useful, how would you propose sharing it?

lonedevwolf, 2017-07-04:
I managed to do it like this 

   camera0.startCapture()
    frame0.queueFrameCapture()
    print "waiting for ext trigger"
    while(camera0.SyncInLevels == 0):
        time.sleep(0.001)
    frame0.waitFrameCapture()
and by setting the camera to execute 
AcquisitionStart and AcquisitionStop on Rising and Falling edge of SyncIn respectivly,

Since you probably need only network access to the camera i think it would be best to use some kind of VPN or i can set up a PC and give you ssh access. Tell me what you need and i will take care of it, because your wrapper has cut my work load in half

CrocodileDandy, 2017-08-22:
@lonedevwolf : Hey, I came across your problem while looking up the different camera features and how frame rate is managed. [This application note](https://www.alliedvision.com/fileadmin/content/documents/products/cameras/various/appnote/various/Migrating-to-Goldeye-G-from-earlier-Camera-Models.pdf) may interests you. It concerns Goldeye G cameras and how to handle the transition to those new cams if you own older ones of the same type. Though it describes frame rate management and I think it applies to your Mako G030B. Then just use Pymba to access the features you need.