morefigs/pymba #51: ImportError with vimbastructure (and a solution proposal)

CrocodileDandy, 2017-08-03:
Hi,

I currently use Pymba within the following directory structure:
`Instruments
        |------ some files we don't care about 
        |------ Cameras
                          |------ Camera_AVT.py
                          |------ pymba`
where:
- the `pymba` folder is the wrapper main folder,
- I have not run pymba's `setup.py` because I did not want to install anything on the local Python distribution
- `Camera_AVT.py` contains a camera class using the wrapper through a `import pymba`,
- the parent folder of `Instruments` is in my PYTHONPATH (yes, Windows 10 here),
- and I use a script located somewhere else that imports `Camera_AVT.py` through a `from Instruments.Cameras.camera_AVT import Camera_AVT`

Now when I do that, I get an error while importing `vimbastructure.py` in `vimba.py` (and elsewhere):
`Traceback (most recent call last):
  File "path_to_my_script/my_script.py", line X, in <module>
    from Instruments.Cameras.camera_AVT import Camera_AVT
  File "some_path\Instruments\Cameras\camera_AVT.py", line 10, in <module>
    import pymba
  File "some_path\Instruments\Cameras\pymba\__init__.py", line 3, in <module>
    from .vimba import Vimba
  File "some_path\Instruments\Cameras\pymba\vimba.py", line 3, in <module>
    import pymba.vimbastructure as structs
ImportError: No module named pymba.vimbastructure`

So I brushed up my understanding of Python's import logic and understood this is because Python fails a absolute/relative import due to the fact that the `pymba` parent folder is not in the PYTHONPATH (or the PATH) when I try to perform a `import pymba.vimbastructure as structs`.

I changed all occurrences of `import pymba.vimbastructure as structs` for `from . import vimbastructure as structs` in order to perform a proper relative import according to [PEP238](https://docs.python.org/2/whatsnew/2.5.html#pep-328-absolute-and-relative-imports). It works.
So is there a smart reason the import was made with `import pymba.vimbastructure as structs` rather than `from . import vimbastructure as structs`  and the way I use the package is just not proper ?
If not, I would like to ask permission to modify this.

FYI, `import pymba.vimbastructure as structs` occurs in:
`vimba.py
vimbacamera.py
vimbadll.py
vimbafeature.py
vimbaframe.py
vimbafeature.py
vimbainterface.py
vimbaobject.py`

morefigs, 2017-08-03:
Yes, changing to `from . import vimbastructure as structs` should be fine, thank you.

CrocodileDandy, 2017-08-10:
I cloned, created a new branch with an adapted name, made my change, committed, tried to push my branch, got kicked out by Github, switched to SSH authentication, put my public key on my profile, still getting kicked out.
I assumed I had to do that in order to create a PR. What am I doing wrong ?

morefigs, 2017-08-10:
Hard to diagnose from here. The easiest way is probably to fork it on github, make the changes to your forked repo, then submit a PR.

CrocodileDandy, 2017-08-12:
Done.

closed 2017-08-13 by CrocodileDandy