morefigs/pymba #53: GUI

robmarkcole, 2017-08-16:
Has anyone made a GUI for pymba? e.g. using pyQT, wxPython

derricw, 2017-08-16:
You mean like a gui with live view?  I made one at work using [pyqtgraph](http://www.pyqtgraph.org/) fairly easily.

robmarkcole, 2017-08-16:
Precisely. Would like to do live background subtraction whilst doing alignment. 

Any chance your code is on GitHub?

morefigs, 2017-08-16:
I've previously made GUIs for pymba using PyQt for controlling the camera and OpenCV to display images in real time, which worked well. I don't have the code available to share unfortunately.

robmarkcole, 2017-08-21:
@derricw any chance you can post your code?
Cheers

jdahlen0441, 2018-02-01:
Here is an example GUI that I wrote using tkinter that displays a live video from an AVT camera using the pymba library.  

I am using a USB to Ethernet connector and a 235 camera.  Both color and mono works fine.

I am not a professional programmer so if others could improve on this code that would be great.  

Thanks to morefigs for making the library available, I really appreciate it.

I have organized the code into two files, the first being the AVT camera class:
GIGECamera_2_1_18.py

The second being the tkinter class:
Tkinter...py

Camera Class:

```

#Import libraries
import pymba #vimba package - Allied Vision Camera
import time #included in python - time delays
import numpy as np #numpy package - matrix
import cv2 #Used if bayerRGB format is selected to convert format into something tkinter can use

class GIGECamera():
    def __init__(self):
        vimba = pymba.Vimba()    
        vimba.startup()
        system = vimba.getSystem()
        system.runFeatureCommand("GeVDiscoveryAllOnce")
        time.sleep(0.2)
        
        try:
            camera_ids = vimba.getCameraIds()
            for cam_id in camera_ids:
                print("Camera found: ", cam_id)
                
            self.cam = vimba.getCamera(camera_ids[0])
            self.cam.openCamera()
            
            #Set Single Frame or Continuous operation
            self.cam.AcquisitionMode='SingleFrame'
            #self.cam.AcquisitionMode='Continuous'
            print(self.cam.AcquisitionMode)
                            
            #self.cameraFeatureNames = self.cam.getFeatureNames()
            #for name in self.cameraFeatureNames:
            #    print('Camera feature:', name)
            
            #Set pixel format
            #Mono formats for 235
            #Mono8, Mono12Packed, Mono12
            #Color formats for 235
            #Mono8, BayerRG8, BayerRG12, RGB8Packed, BGR8Packed, YUV411Packed, YUV422Packed, YUV444Packed            
            self.cam.PixelFormat="Mono8"
            #self.cam.PixelFormat="RGB8Packed"
            #self.cam.PixelFormat="BayerRG8"
                
            #If we use the Bayer format - the speed is 4-5x more than RGB8Packed
            #https://docs.opencv.org/3.1.0/de/d25/imgproc_color_conversions.html
            #cv::COLOR_BayerBG2BGR, cv::COLOR_BayerGB2BGR, cv::COLOR_BayerRG2BGR, cv::COLOR_BayerGR2BGR, cv::COLOR_BayerBG2RGB, cv::COLOR_BayerGB2RGB, cv::COLOR_BayerRG2RGB, cv::COLOR_BayerGR2RGB 
                               
            #Set Package size
            #The default is 8228 - but 1500 works with either the USB 2 or 3 adapter
            #Could not get anything higher than 1500 to work
            print(self.cam.GevSCPSPacketSize)
            self.cam.GevSCPSPacketSize=1500
            print(self.cam.GevSCPSPacketSize)
                
            #Set Payload Size
[... 153 lines cut ...]
                except:
                    pass
                    
if __name__ == "__main__":
    root = tk.Tk()
    root.geometry("1500x700")
    root.attributes('-topmost',True)
    root.after_idle(root.attributes,'-topmost',False)
    root.wm_title("Tkinter AVT GUI") #Makes the title that will appear in the top left
    
    #Setup the AVT camera
    camera = GIGECamera()
    time.sleep(0.2)
    
    #load main application
    MainApplication(root).pack(side="top", fill="both", expand=True)
    root.mainloop()
    
    #Close Camera
    try:
        camera.close() 
    except:
        pass

```