morefigs/pymba #67: delay issues on USB AVT Cam mako U-130

Toulmin, 2018-05-16:
Hello,
We are developing a software able to perform marker tracking on live images. we were previously using Giga Ethernet AVT Cam manta G917 without any problem.
In order to increase working frequency, we bought a USB AVT Mako U-130, we have been able to use our software with this new camera but we encountered a delay issue.
This delay between an event and its display seems to increase in time and can reach as high as 20 seconds.

Can we find documentation or code examples using pymba with USB camera ?
Are there important differences of coding between USB and Giha Ethernet cameras ?

For information :
- Our OS is Windows 10
- We don't have any problem with Vimba Viewer interface (no delay).
- We use Qt interface to display images with our software.
- We use python 2.7
- We are using the last version of pymba and vimba 2.1.3.

morefigs, 2018-05-17:
Do you know if the delay is in Python acquiring the image data, or in Qt displaying it? It may be simply a matter of Qt not being updated to draw the image.

Other than that I haven't had much experience using USB cameras but others around here may have. There's not much documentation for Pymba, but it is essentially just a wrapper around Vimba, so the Vimba documentation might have your answer. It may be that extra and/or different function calls are recommended when using USB.

Toulmin, 2018-05-17:
Thank you for your message.
To me this is a python issue.
With Qt, we only translate a numpy array image to a QImage and display it. There is no reasons that those actions would take more time with a USB camera than a Giga Ethernet camera.

Unfortunately I haven't found a lot of documentation on vimba either.

morefigs, 2019-01-30:
Is this still an issue?

closed 2019-02-07 by morefigs