morefigs/pymba #68: Use ExposureAuto and GainAuto

PedroSilva0, 2018-05-21:
I would like to adapt the live display example to use AutoGain and AutoExposure in "Continuous" mode, but when I set those features to  "Continuous" the live display doesn't adapt to lighting changes. 
I would like to know if this is a limitation of Pymba or if I need to do extra modifications instead of only changing those two features values to "Continuous"?

morefigs, 2018-05-22:
There are a few different parameters that can affect e.g. automatic gain, see page 64 and onwards here: https://www.alliedvision.com/fileadmin/content/documents/products/cameras/various/features/GigE_Features_Reference.pdf.

It's been a while, but you should be able to set those sorts of things by setting camera attributes.

GavinJia, 2018-10-25:
I tried to set as below:
```
self.camera = self.vimba.getCamera(self.cameraIds[0])
self.camera.openCamera()
self.camera.AcquisitionMode = 'Continuous'
self.camera.ExposureAuto = 'Continuous'
```
Two situations were happened. Overexposure or underexposed.
I don't know why...

closed 2019-01-30 by morefigs