morefigs/pymba #69: Package installation problems

pfefferminza, 2018-05-30:
Hi, 

I'm a bit confused as to how to get started with pymba (I'm not super experienced with python yet). Here're the steps I followed:

1. I downloaded the lastest pymba-master folder and navigated using command prompt to this folder
2. I ran 'python setup.py install'

This created following folder: 'C:\Python27\Lib\site-packages\pymba-0.1-py2.7.egg'
within 'pymba-0.1-py2.7.egg' are now two folders: 'EGG-INFO' and 'pymba'.

3. I opened a python shell and tried running the command 'import pymba'

This created following error: 'ImportError: No module named pymba'.

4. I then moved the the location of the 'pymba' folder to be directly inside 'C:\Python27\Lib\site-packages'

This got rid of the error, however a new error arose: 'IOError: VimbaC.dll not found'

5. I then changed the path where to find VimbaC.dll inside vimbadll.py to point to where I can find VimbaC.dll in my vimba folder: 'r'C:\Program Files\Allied Vision\Vimba_2.1\Tools\Viewer\Win64\VimbaC.dll'

Now I'm getting a new error to do with importing vimbastructure: 

  File "<pyshell#2>", line 1, in <module>
    import pymba
  File "C:\Python27\lib\site-packages\pymba\__init__.py", line 3, in <module>
    from .vimba import Vimba
  File "C:\Python27\lib\site-packages\pymba\vimba.py", line 3, in <module>
    from . import vimbastructure as structs
ImportError: cannot import name vimbastructure

I'm stuck with this error since vimbastructure.py seems to be inside the pymba folder and I'm not sure why it's having trouble importing it.

Does anyone know how to fix this? Or am I doing something completely wrong when setting the package up? I've also tried 'pip pymba', however that gave me an error telling me that I'm missing Visual C++.

Thanks for your help!
Pfefferminza

anki-xyz, 2018-06-01:
As I wrote on my website, it is maybe a conflict of your Python version, that is at least the Visual C++ issue pointing to.  The last problem you are describing was described and "solved" here: #51 
My advise: Changing the Python environment or pulling an older version of pymba.

pfefferminza, 2018-06-01:
Thanks a lot for your reply! 
I've tried the 2017 version of the line: import pymba.vimbastructure as structs, which unfortunately is giving me the same error :/ I've also switched to anaconda to test a new python environment: the same error. 
Do you still think switching to python 3 could help?

morefigs, 2019-01-30:
If you use Pymba v0.2 and Python 3 you can simply install via `pip install pymba`. This may help your installation issues.

closed 2019-02-04 by morefigs