morefigs/pymba #70: Pymba doesn't work without Vimba

sliusar, 2018-07-08:
Hi all,

I have an issue using pymba with Prosilica GT1380 and GT3300 cameras: the example "Interacting with cameras" from README.md when executed (after cameras were power cycled) returns array which has correct dimensions depending on the camera resolution but all values are 0, to fix that I simply need to start Vimba viewer and click start button for the camera to get a single frame, then I close Vimba viewer and the example script gets a perfect picture every time when executed. Could you please help to solve the issue? I use the most recent pymba from here.

Thanks for any help.

sliusar, 2018-07-08:
I've compared the features values right after power cycle and after getting single frame with Vimba viewer. The issue was in GVSPPacketSize and GevSCPSPacketSize, both values were 8228, the Vimba viewer set them to 1500, so now I change this features for camera0 and script works without Vimba.

Sorry for bothering.

closed 2018-07-08 by sliusar

seahawk67, 2018-07-08:
Another option would be to enable Jumbo Frames on your network card (if supported).

sliusar, 2018-07-09:
Unfortunately in current setup it is impossible due to some PLCs situated on the same switch where moving to jumbo is not that trivial, but eventually we might consider that, thanks for tip.