morefigs/pymba #72: Capture of multiple frames

ohinrichs, 2018-09-05:
In a test construction we try to captureframes with a very high framerate. The camera is taking pictures behind a strobe, which is turning with frequency of about 50 Hz and has a small window. The signal von the strobe can be used as an external trigger. So the plan is to start taking multiple frames (about 300) after the external trigger (maybe with some delay) with a high framerate (about 2000 to 3000 fps) and then saving the frames to a file. When the trigger is disabled the framerate is achieved. But when the trigger is activated  only 25 fps are possible, even when "cam.AcquisitionFrameCount = 300". We tried different combinations of TriggerSelector (AcquisitionStart, AcquisitionRecord, FrameStart) and AcquisitionMode (Continuous, Multiframe, Recorder).
It seems that only one frame is capture or save after the trigger. 
So the question is: Do we have to use another function to get multiple frames after the trigger than "frame.queueFrameCapture()" and "frame.getBufferByteData()"?
How does the recorder work? Can you provide an example for capturing multiple frames at once?

morefigs, 2018-09-17:
I don't know off the top of my head. It's probably more a question of reading the Vimba manual and seeing if there is functionality for high speed capture, as generally speaking pymba is just a thin wrapper for Vimba functions.

sciunto, 2018-09-26:
I do have the same question. Indeed, the online [documentation ](https://www.alliedvision.com/fileadmin/content/documents/products/cameras/various/features/Camera_and_Driver_Attributes.pdf) mentions various modes of acquisitions, which are not currently illustrated in the examples or the documentation of pymba.

sciunto, 2019-02-08:
thanks @morefigs !

morefigs, 2019-02-09:
There's good info in the Vimba manual about doing the above.

morefigs, 2019-02-11:
@ohinrichs I've added camera convenience functions for streaming images asynchronously, which should be an efficient way to capture at high frame rates. You may need to tweak `frame_buffer_size` in order to be able to save frames to disk at such a high frame rate without dropping frames.

morefigs, 2019-02-17:
Closing for now. If this is still an issue with Pymba >= 0.3 please open a new issue.

closed 2019-02-17 by morefigs