morefigs/pymba #76: Can not show image in Python

GavinJia, 2018-10-25:
If I open the camera via Vimba Viewer first, it can be opened via python.
It is workable and shows image.
If I open the camera directly via python, it can't be opened.

Do you have any idea for this?

```
start...
openCamera...
The Vimba lib version is: 1.7.0 , and vimba run successfully!
Available cameras ['DEV_000F315C1E2D']
[AcquisitionMode] Continuous
[ExposureAuto] Continuous
[ExposureAutoAdjustTol] 5
[ExposureAutoAlg] Mean
[ExposureAutoMax] 10000
[ExposureAutoMin] 100
[ExposureAutoRate] 100
[ExposureAutoTarget] 40
[Init_ExposureTimeAbs] 9998.0
[Init_Gain] 0.0
showImage...
[Exposure] 9998.0
[Gain] 0.0
captureFrame...
Traceback (most recent call last):
  File "D:\AVT_test.py", line 158, in <module>
    main()
  File "D:\AVT_test.py", line 147, in main
    ins.start()
  File "D:\AVT_test.py", line 137, in start
    self.captureFrame()
  File "D:\AVT_test.py", line 105, in captureFrame
    self.frame0.queueFrameCapture()
  File "C:\Python35\lib\site-packages\pymba-0.1-py3.5.egg\pymba\vimba_frame.py", line 126, in queueFrameCapture
    raise VimbaException(errorCode)
pymba.vimba_exception.VimbaException: Other error.
```

carasuca, 2018-11-29:
@GavinJia same problems here. Are you using a GigE camera? The viewer might initialize something related to MTU / bandwidth. Try this:
```python
camera0 = vimba.getCamera(cameraIds[0])
camera0.runFeatureCommand('GVSPAdjustPacketSize')
```

morefigs, 2019-01-30:
You could compare differences in feature values with the camera started before and after Vimba Viewer. There are 3 files that output feature values:

https://github.com/morefigs/pymba/blob/master/examples/camera/list_feature_values_and_ranges.py
https://github.com/morefigs/pymba/blob/master/examples/interface/list_feature_values_and_ranges.py
https://github.com/morefigs/pymba/blob/master/examples/system/list_feature_values_and_ranges.py

carasuca, 2019-02-10:
@morefigs setting low `GevSCPSPacketSize` and `StreamBytesPerSecond` fixed it for me. I haven't tried the scripts. Unfortunately I do not have access to the hardware anymore.

closed 2019-02-10 by morefigs