morefigs/pymba #77: Unable to run gige vision mako camera...

chandanlal92, 2018-12-07:
I am getting following error if I run opencv_example.py

 camera0 = vimba.getCamera(cameraIds[0])
IndexError: list index out of range

and also test_camera.py

chandanlal92, 2018-12-07:
Please help to configure mako-G 503 Gige Vision Camera using pympa....I am unable to connect the camera

chandanlal92, 2018-12-07:
is there any changes to be done for Gige Vision Camera in the Code

morefigs, 2019-01-28:
Are you calling `system.runFeatureCommand("GeVDiscoveryAllOnce")`?

chandanlal92, 2019-01-28:
Yes I am calling system.runFeatureCommand("GeVDiscoveryAllOnce")

morefigs, 2019-01-28:
Please try adding a few second wait (`time.sleep(10)`) after calling the above command to see if that helps.

morefigs, 2019-01-30:
You could try Pymba v0.2.

chandanlal92, 2019-01-30:
Tried both the versions with delay still not same error.

chandanlal92, 2019-01-30:
For GigE Camera is there anyother thing that needs to be changed?

morefigs, 2019-01-30:
Does the camera work if you open it in Vimba Viewer? 

https://github.com/morefigs/pymba/blob/master/examples/camera/list_camera_ids.py works for me with a GigE camera and Pymba v0.2.

Could you also try https://github.com/morefigs/pymba/blob/master/examples/interface/list_interface_ids.py to see if you can see an interface?

Also try running the above examples after opening the camera in Vimba Viewer.

chandanlal92, 2019-01-31:
I am able to work with the camera and in vimba viewer but it is not being listed in the camera list in any of the interfaces like vimbaC,Vimba C++ and Pymba

morefigs, 2019-02-07:
Did that help? Is this still an issue?

chandanlal92, 2019-02-08:
No I still have the issue.It is not solved.

chandanlal92, 2019-02-08:
What IP Config did you use Persistant or LLA?

morefigs, 2019-02-09:
I imagine you don't ever want to be using LLA. DHCP is probably preferable for avoiding networking issues. I'm using DHCP.

chandanlal92, 2019-02-15:
I am able to stream using pymba now. The problem got resolved once I removed firewall. Thank you for your support will let you if there is any issues in future.

closed 2019-02-15 by chandanlal92