morefigs/pymba #81: [Mako U-130] delayed frame acquisition

tabatabai, 2019-02-25:
Thank you for the work you put into pymba, it has been really helpful.

On the current version of pymba, trying the example opencv_acquire_streaming_images.py I noticed that the live view is delayed by roughly a second.

Using an older version of pymba the same thing happens. There my code looks roughly as follows:

```javascript
frame = camera.getFrame()
frame.announceFrame()
camera.startCapture()
camera.runFeatureCommand('AcquisitionStart')

while True: # this loop actually gets controlled by a hardware switch
    frame.queueFrameCapture()
    frame.waitFrameCapture()
    frame_data = frame.getBufferByteData()
    data_frame = np.ndarray(buffer=frame_data,
                            dtype=np.uint8,
                            shape=(frame.height, frame.width))
    queue.put(np.copy(data_frame)) # putting the data into a queue for further processing

camera.runFeatureCommand('AcquisitionStop')
camera.endCapture()
vimba.shutdown()
```

Pausing and resuming the loop it becomes clear that the obtained data lags behind.
For example:
1. starting the capture loop
2. pointing the camera at object A for a few hundred iterations (that's just a few seconds on the Mako),
3. pausing the capture loop
4. pointing the camera at object B
5. resuming the capture loop
6. the first 50-150 images will still be of object A

This behaviour is the same as when running the live view from opencv_acquire_streaming_images.py.

I would be grateful for any pointers to fix this.

Thank you for your work and time,
Paul

morefigs, 2019-02-25:
Strange. I would've thought that by only announcing (`frame.announce()`) a limited number of frames then that would limit the buffer size that the camera uses. By default Pymba's `arm` method only announces 10 frames.

If your camera is indeed buffering more data internally then there may be a camera feature that you can change to prevent that. Running https://github.com/morefigs/pymba/blob/master/examples/camera/list_feature_values_and_ranges.py may help identify any relevant features.

tabatabai, 2019-02-26:
Thank you for your response.

I suspect that setting StreamBufferHandlingMode to 'NewestOnly' would fix the issue. While this option appears in the [GenICam GenTL SFNC 1.1.1](https://www.emva.org/wp-content/uploads/GenICam_GenTL_SFNC_1_1_1.pdf), in the [Mako's Features Reference](https://www.alliedvision.com/fileadmin/content/documents/products/cameras/various/features/Mako_U_Features_Reference_V1.1.0.pdf) the only valid value is 'Default' and I can't change StreamBufferHandlingMode in either VimbaViewer or via pymba because it appears to be locked.

morefigs, 2019-02-27:
Do you get the same behaviour in Vimba Viewer?

When you `open()` a camera it's opened with the default access mode of `VMB_ACCESS_MODE_FULL`. You could try other values e.g. `VMB_ACCESS_MODE_CONFIG`. I've not tried that before so not sure how safe it is.

tabatabai, 2019-02-27:
There is no delay in Vimba Viewer. Opening the camera in CONFIG mode does unfortunately not work: `VimbaException: API feature is not implemented.`

morefigs, 2019-03-04:
Vimba Viewer has a log window (on the right side). Does this list any settings that are being changed when you open/run the camera?

tabatabai, 2019-03-11:
No, unfortunately the log window does not show any changes.

morefigs, 2019-03-13:
Unfortunately I don't have a USB camera to test on. Vimba Viewer uses the C++ API I believe, so it may be setting something not possible in Vimba C. The Vimba docs might prove useful.

tbenst, 2019-04-12:
Same issue for me with U-130, I contacted Allied Vision and they are looking into it...will update here as I have info

morefigs, 2019-04-27:
Is this still an issue? Or is it due to camera hardware, etc?

tabatabai, 2019-07-04:
Apologies for the late reply. I solved the issue by lowering the acquisition frame rate from 168 to 144. I think the problem was that the python loop was running with less than 168 iterations per second, which led to images being buffered.

Thank you for your time and help!
Paul

morefigs, 2019-07-04:
Great :)

closed 2019-07-04 by morefigs

albertoSantana28, 2021-08-06:
> Apologies for the late reply. I solved the issue by lowering the acquisition frame rate from 168 to 144. I think the problem was that the python loop was running with less than 168 iterations per second, which led to images being buffered.
> 
> Thank you for your time and help!
> Paul

Paul, I have the same problem as you, when you say you fixed the problem by lowering the acquisition frame, do you mean that this setting is in the camera? did you change it with the vimba software?

Regards