morefigs/pymba #82: License change question

adam-urbanczyk, 2019-03-06:
I noticed that this (very useful!) project changed license to the more restrictive GPL3. Any particular reason to do so?

morefigs, 2019-03-07:
Pymba originally used a GPL license, but was changed briefly to an MIT license: https://github.com/morefigs/pymba/commits/master/LICENSE. It's now been changed back to GPL. Happy to discuss further via email if you have specific needs.

adam-urbanczyk, 2019-03-07:
Do you have anything against using and developing the older version under the previous license, e.g. as  forked here https://github.com/adam-urbanczyk/pymba ?

morefigs, 2019-03-07:
Not at all. But if you're using Python 3 then Pymba 0.3+ has had a few improvements. Is the license the only issue stopping you?

adam-urbanczyk, 2019-03-09:
Thanks! Yes, indeed it is the license. BSD or MIT would be more optimal.

morefigs, 2019-03-12:
OK, done.

closed 2019-03-12 by morefigs

adam-urbanczyk, 2019-03-13:
Thanks a lot!