morefigs/pymba #83: Camera object has no attribute GVSPAdjustPacketSize

alexfieg, 2019-03-08:
Hey,

I really appreciate your work. When I tried to run my MAKO U-029B connected via USB using the opencv_acquire_image.py I get the attached error. Any ideas why my camera does not have this attribute?

'''
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-6-6f11d38875a1> in <module>
     23     with Vimba() as vimba:
     24         camera = vimba.camera(0)
---> 25         camera.open()
     26 
     27         camera.arm('SingleFrame')

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pymba\camera.py in open(self, camera_access_mode, adjust_packet_size)
    134         # may experience issues with camera comms if not called
    135         if adjust_packet_size:
--> 136             self.GVSPAdjustPacketSize()
    137 
    138     def close(self):

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pymba\vimba_object.py in __getattr__(self, item)
     35             return feature.value
     36 
---> 37         raise AttributeError(f'{self.__class__.__name__} object has no attribute {item}')
     38 
     39     # allow direct access to feature values as an attribute

AttributeError: Camera object has no attribute GVSPAdjustPacketSize
'''

morefigs, 2019-03-08:
Oops that's a bug. For now, you can change `camera.open()` to `camera.open(adjust_packet_size=False)`. In the meantime I'll fix it.

alexfieg, 2019-03-08:
works, thanks!

morefigs, 2019-03-09:
Fixed in v0.3.2.

closed 2019-03-09 by morefigs