morefigs/pymba #84: Vimba other Error whilst capturing series  of images

LukasRosenbauer, 2019-03-08:
Hi,

I am capturing Images with a Mako G-503C in Series using Pymba 0.1. After a (seemingly random) amount of Images I get the following error:
> pymba.vimbaexception.VimbaException: Other error.
whilst calling announceFrame() method.
I use the following source Code:
`"""This module provides all functionalities regarding the camera access"""

import pymba
import numpy
import cv2
import time
import os

ImageDirectory = ""
ImageName = ""
Camera = None
Vimba = None
ID = "DEV_000F315C80FF"

def _Init():
    """This function executes all defined jobs for setting up the module"""
    pass

def _SetImageLocation(Directory, Name):
    """
    This method sets the image location based on the specific directory and image name.

    :param Directory: Directory to set - type "str" (e.g. "C:\\temp")
    :param Name: Name to set (without file ending!) - type "str" (e.g. "MyPicture")
    """
    global ImageDirectory
    global ImageName
    ImageDirectory = Directory
    ImageName = Name

def _GetImageLocation():
    """
    This method gets the image location.

    :return: Image location - type "str) (e.g. "C:\\temp\\MyPicture")
    """
    global ImageDirectory
    global ImageName
    return os.path.join(ImageDirectory, ImageName + ".jpg")

def GetCamera():
    '''
    Applies the Singleton pattern to create unique vimba and camera instances.
    Thus those are only once created.
    '''
    global ID
    global Camera
    global Vimba
    if Camera is None:
        # Create vimba instance
        Vimba = pymba.Vimba()
        # Startup vimba
        Vimba.startup()
        # Get system object
        System = Vimba.getSystem()
        # Discover cameras
        if System.GeVTLIsPresent:
            System.runFeatureCommand("GeVDiscoveryAllOnce")
            time.sleep(0.200)
        # Assign camera
        Camera = Vimba.getCamera(ID)
        # Open camera
        Camera.openCamera()
        # Set acquisition modes
        Camera.AcquisitionMode = "Continuous"
    return Camera

def Shutdown():
    '''
    tears down the component
    '''
    global Vimba
    if Vimba is not None:
        Vimba.shutdown()

def Snapshot():
    '''
    Uses the global camera object to generate a new image.
    '''
    Camera = GetCamera()
    # Start capture
    Camera.startCapture()
    # Assign frames
    Frames = [Camera.getFrame() for _ in range(1)]
    # Iterate each frame
    for Frame in Frames:
        # Announce frame
        Frame.announceFrame()
        # Queue frame capture
        Frame.queueFrameCapture()
    # Acquisition
    Camera.runFeatureCommand("AcquisitionStart")
    time.sleep(0.500)
    Camera.runFeatureCommand("AcquisitionStop")
    # Wait for frame buffer
    Frame.waitFrameCapture()
    # Convert image
    Image = numpy.ndarray(buffer=Frame.getBufferByteData(),
                          dtype=numpy.uint8,
                          shape=(Frame.height, Frame.width, 1)
                          )
    Image = cv2.cvtColor(Image, cv2.COLOR_BAYER_GR2RGB)
    Image = cv2.cvtColor(Image, cv2.COLOR_BGR2GRAY)
    # Save image
    cv2.imwrite("C:\\Temp\\test.jpg", Image)
    # Cleanup
    Camera.endCapture()
    Camera.revokeAllFrames()

counter = 0
while True:
    counter = counter + 1
    Snapshot()
    print(counter)

`

morefigs, 2019-03-09:
You can't announce a frame after capture has started. I'm surprised it works for any cameras. Swapping start capture and announce frame functions should help.

I'd strongly recommended downloading Pymba 0.3.2 and trying the examples there, which are much simplified.

LukasRosenbauer, 2019-03-11:
Thanks that solved the problem

closed 2019-03-11 by LukasRosenbauer