morefigs/pymba #87: Timeout when capturing multiple frames

vascotenner, 2019-03-27:
I have pymba implemented in a exisiting GUI application. However, it crashes frequently withouth error message. I could repoduce this behaviour partially while running the following code under linux, pymba-master and a GigE camera:

    with Vimba() as vimba:
        camera = vimba.camera(0)
        camera.open()

        # capture a single frame, more than once if desired
        for i in range(10000):
            camera.arm('SingleFrame')
            frame = camera.acquire_frame()
            print('Image {}, mean {}'.format(i, frame.buffer_data_numpy().mean()))
            camera.disarm()
        camera.close()

Now it gives the following error infrequently after some frames (between 10 and 5000):
```
Traceback (most recent call last):
  File "opencv_acquire_image.py", line 15, in <module>
    frame = camera.acquire_frame()
  File "/home/vtr300/src/pymba_morefigs/pymba/camera.py", line 236, in acquire_frame
    self._single_frame.wait_for_capture()
  File "/home/vtr300/src/pymba_morefigs/pymba/frame.py", line 95, in wait_for_capture
    raise VimbaException(error)
pymba.vimba_exception.VimbaException: Timeout during wait.
```

I am aware that arming and disarming the camera might not be the cleanest way of coding.

morefigs, 2019-03-28:
Do you need to arm and disarm within the loop? I also get the crash if I do it that way, but not if I arm and disarm outside the loop. It's also slower of course.

tomasstankevic, 2019-04-03:
I am having the same issue whe **not** arming and disarming in every iteration. It was not there in version 0.1

morefigs, 2019-04-07:
In version 0.1 the lines
    
    self._single_frame.wait_for_capture(timeout_ms)
    self.AcquisitionStop()

were reversed. The above _should_ be correct, but you could try calling `self.AcquisitionStop()` before waiting for capture to see if that helps. I'm curious to see if it does.

Alternatively you can try catching `VimbaException`s and ignoring them.

vascotenner, 2019-04-11:
Is there any reason to disarm the camera?

morefigs, 2019-04-12:
No, you shouldn't need to arm/disarm the camera every frame.

morefigs, 2019-04-13:
As far as I can tell this is normal behaviour.

The problem may be exacerbated when using longer exposure times. Note that timeout can be increased with `camera.acquire_frame(timeout_ms)`, which may help.

I've updated the example to catch and ignore timeout errors.

closed 2019-04-13 by morefigs

vascotenner, 2019-04-13:
Can you comment on any usecase that requires the use to disarm?

morefigs, 2019-04-13:
Arming/disarming starts and stops the capture engine and frame acquisition. I don't know explicitly but it's probably only required if you want to change certain camera settings.