morefigs/pymba #88: Operation is invalid with the current access mode

PereeT, 2019-03-28:
I'm trying to use pymba for my new camera Mako U-130-B (USB3.0) and i got the message : "VimbaException: Operation is invalid with the current access mode". The error only appears if I write : camera.PixelFormat = 'Mono8' (or another parameter like this one). If I don't try to specify any parameter I got an image succesfully. 

By reading the camera.py, I tried to write : camera_NIR.open(camera_access_mode = vimba_object.VimbaObject.VMB_ACCESS_MODE_FULL) but it didn't change anything (probably because it's already the default parameter). 

I really tried to search a solution on internet but I can't find any and I'm running out of ideas. Do you have a lead for me ? 

PS : Thank you for the version 0.3.3 it solved the problem of my previous topic as you said it would.

morefigs, 2019-04-01:
And you can set those parameters in Vimba Viewer?

PereeT, 2019-04-01:
Yes I can

morefigs, 2019-04-07:
That's strange. I'll have one of those cameras soon so can test. In the meantime it could be worth reading the Vimba documentation about access modes, etc.

tbenst, 2019-04-12:
Same issue for me with U-130. I restarted my computer and everything worked!

PereeT, 2019-04-15:
It is one of the first thing I tried but it didn't worked.

tbenst, 2019-04-15:
@PereeT I misread your issue. I see the same error if I try, `side_camera.AcquisitionFrameRate = 100` for example

morefigs, 2019-04-23:
@PereeT can you successfuly run https://github.com/morefigs/pymba/blob/master/examples/camera/set_feature_value.py? I have the same camera and the above works for me.

PereeT, 2019-04-23:
Yes this part works and it helped me to go further in my investigation. The problem comes from this line : "camera_NIR.arm('SingleFrame')". If I don't write it, I don't have the message error and I can modify succesfully the parameters. For the parameter "SingleFrame" I used the code from set_feature_value.py

"feature = camera_NIR.feature('AcquisitionMode')
acquisition_mode = feature.value
feature.value = 'SingleFrame'"

This doesn't raise any exception and I can see in the variables that acquisition_mode = 'SingleFrame' but it seems to have raised another problem. If I try to take a frame with this code :

"for j in range(1):
           frame = camera_NIR.acquire_frame()
           print(f'frame {frame.data.frameID}')
           image = frame.buffer_data_numpy()"

I get :"VimbaException: Invalid camera mode for the requested operation."

To summarize, by writing camera.arm('SingleFrame') I block any further parameter modification but I can succesfully take a picture. If I don't write it, I can change the parameters but I can't take a picture and if I try to workaround by writing differently the feature 'AcquisitionMode' I can't either take a picture despite the fact that I can see in the variables that acquisition_mode = 'SingleFrame' 

What I don't understand is why the same code works for the Goldeye camera (ethernet cable) and not for this camera (USB 3.0).

morefigs, 2019-04-26:
Yes, to confirm you should use `arm()` before trying to acquire a frame.

And yes I'd expect you not to be able to change (some) parameters after arming, as they can affect things like frame size, which is set upon arming.

closed 2019-04-26 by morefigs

tbenst, 2019-04-26:
Hm, @morefigs that didn't solve the problem for me. @PereeT could you try?

```python
import pymba
from pymba import Vimba
import numpy as np
import cv2

def display_frame(frame, delay=1):
    """
    Displays the acquired frame.
    :param frame: The frame object to display.
    :param delay: Display delay in milliseconds, use 0 for indefinite.
    """
    print(f'frame {frame.data.frameID}')

    # get a copy of the frame data
    image = frame.buffer_data_numpy()

    # convert colour space if desired
    try:
        image = cv2.cvtColor(image, PIXEL_FORMATS_CONVERSIONS[frame.pixel_format])
    except KeyError:
        pass

    # display image
    cv2.imshow('Image', image)
    cv2.waitKey(delay)

with Vimba() as vimba:
    my_id = vimba.camera_ids()[0]
    camera = vimba.camera(my_id)
    camera.open()
    camera.AcquisitionFrameRate = 100
    
    # arm the camera and provide a function to be called upon frame ready
    camera.arm('Continuous', display_frame)
    camera.start_frame_acquisition()

    # stream images for a while...
    sleep(5)

    # stop frame acquisition
    # start_frame_acquisition can simply be called again if the camera is still armed
    camera.stop_frame_acquisition()
    camera.disarm()

    camera.close()
```
`VimbaException: Operation is invalid with the current access mode.`

morefigs, 2019-04-27:
@tbenst I get the same error. It seems that feature simply isn't settable. Vimba Viewer also doesn't allow it. I can however set other feature values e.g. PixelFormat.

zibbini, 2020-04-02:
For any users still trying to adjust frame-rate, I've only found setting `AcquisitionFrameRateEnable = True` prior to setting the frame-rate works reliably:

``` python
from pymba import Vimba

with Vimba() as vimba:
	cameras = vimba.camera_ids()
	camera = vimba.camera(cameras[0]) # select the first camera
	camera.open()

        # Set frame rate to 30 fps
	camera.feature("AcquisitionFrameRateEnable").value = True
	camera.feature('AcquisitionFrameRateMode').value = 'Basic'
	camera.feature("AcquisitionFrameRate").value = 30

	camera.close()
```

I found that I didn't need to set the mode to 'Basic' on my system to adjust frame rate (camera model: monochrome Alvium 1800 u-500m). But I presume it may be on other systems...?

roche-MH, 2021-05-04:
My problem was solved by adding the line `camera.AcquisitionFrameRate = 100`