morefigs/pymba #89: Python 3.6 needed?

titusjan, 2019-04-01:
First, thank you for making Pymba. 

I got a syntax error while trying Pymba with Python 3.5 because in line 39 of `vimba_object.py` the following line is present:

```python
    raise AttributeError(f'{self.__class__.__name__} object has no attribute {item}')
```

Is Python 3.6 required to run Pymba? If so, could you please mention this in the installation instructions? If not, can you please make it work on Python 3.5?

morefigs, 2019-04-13:
Fixed in 0.3.4.

closed 2019-04-13 by morefigs

titusjan, 2019-11-19:
The `vimba_pixelformat.py` module uses `IntFlag`, which is a Python 3.6 addition. So the same question applies: can you either make it work on Python 3.5, or document that Python 3.6 is required? Thanks in advance.

I recommend to use the lowest supported Python version during development to prevent unsupported language features like this from slipping in the code.