morefigs/pymba #90: Stripes in images from old frames

tomasstankevic, 2019-04-02:
Hi,
thanks for developing this package, it comes extremely handy!
I am having one issue when taking multiple frames that the frames often consist of stripes containing parts of previous frames. In this case I was quickly changing illumination so the overall brightness was changing from bright to dark from frame to frame
[image]
The reason there are dark and bright stripes in the image above is because they must have come from different frames.
I have noticed that the issue is more severe the more heavy processing is taking place. I can see these stripes using your simple example of continuous acquisition as well as single frame acquisition. Tested with several monochrome Mako cameras. Vimba Viewer is not showing anything like that. 
Thanks and let me know if you need help debugging this issue

morefigs, 2019-04-07:
Do the stripey frames have different (to normal frames) values for `frame.data.receiveStatus` or `frame.data.receiveFlags`? Does increasing the frame buffer size e.g. `camera.arm('Continuous', display_frame, frame_buffer_size=100)` help?

tomasstankevic, 2019-04-13:
I noticed that the appearance of such incomplete/overlapping frames coincides with Vimba Viewer counting Shoved frames in Stream statistics. This means that Vimba Viewer is filtering these faulty frames. 
Increasing the buffer size did not help. I will try checking the` frame.data.receiveStatus `and `frame.data.receiveFlags` when I get a chance.
I believe the reason for missing data is in the GigE interface since I managed to get rid of the problem when i connect the **camera directly to the PC** network card, not going through the network switch. For some reason when the camera is connected through the network performance becomes dependent on CPU load and I start getting incomplete frames. Perhaps I need to talk to the network engineer to configure the switch in some special way

morefigs, 2019-04-26:
How'd you go with this? I'd be curious to know if we can also filter the bad frames.

morefigs, 2019-05-31:
Closing, as some quick tests show ignoring frames with `frame.data.receiveStatus == -1` fixes the issue.

closed 2019-05-31 by morefigs