morefigs/pymba #91: [Mako 030B] Mono12Packed - wrong data in buffer_data_numpy

ivanrajkovic, 2019-04-10:
Hi, 
Mako 030B has 644x482 pixels. If I read a frame with Mono12Packed mode it should give me 644x482*1.5 uint8 that I can convert to 644x488 uint16 (12 bit dept).
frame.buffer_data() contains correct number of bytes and I can convert that to a 16bit image. if I try to use frame.buffer_data_numpy() I find that it has 644x488 uint8, and that is just first 2/3 of the frame.buffer_data. It is impossible to recreate full 12bit depth image from it.
If you need I can send you examples of frame data.

Ivan

ivanrajkovic, 2019-04-10:
I just saw that this is already marked as to-do in the code.

closed 2019-04-10 by ivanrajkovic

morefigs, 2019-04-12:
It's on the todo list, but feel free to add a pull request too.