morefigs/pymba #92: Unhandled Exceptions in Ubuntu for ARM

morphogencc, 2019-05-01:
I'm running the same python script in both Windows 10 and Ubuntu 16.04 for ARM, using Python 3 and Pymba; however, it runs fine indefinitely on Windows, but an exception causes errors and failures after one cycle on Ubuntu.

The code (based off of the `opencv_acquire_image.py` example):

```
def capture_frame(camera):
    frame = None
    camera.arm('SingleFrame')
    # capture a single frame, more than once if desired
    print("Arming Camera for capture")
    try:
        frame = camera.acquire_frame(timeout_ms=5000)

    except VimbaException as e:
        if e.error_code == VimbaException.ERR_TIMEOUT:
            print("VIMBA: {}".format(e.message) )
            camera.disarm()
            camera.arm('SingleFrame')
        else:
            print("VIMBA: {}".format(e.message) )
            raise

    if frame:
        save_frame(frame)
        
    camera.disarm()

def save_frame(frame):
    image = frame.buffer_data_numpy()

    # convert colour space if desired
    try:
        image = cv2.cvtColor(image, PIXEL_FORMATS_CONVERSIONS[frame.pixel_format])
    except KeyError:
        pass

    # save image
    print("Saving image")
    cv2.imwrite('images/{}.png'.format(time.strftime("%Y%m%d-%H%M%S")), image)

if __name__ == '__main__':
    create_directory("images")

    with Vimba() as vimba:
        print(vimba.camera_ids())
        available_cameras = vimba.camera_ids()

        camera = vimba.camera(0)
        try:
            camera.open()
        except VimbaException as e:
            print("VIMBA: Camera unavailable for connection.  Exiting program.")
            quit()

        while True:
            capture_frame(camera)

        camera.close()
        print("Closing camera")
```

The error I'm seeing is:

```
Arming Camera for capture
Saving image
Arming Camera for capture

Traceback (most recent call last):
  File "capture.py", line 25, in capture_frame
    frame = camera.acquire_frame(timeout_ms=5000)
  File "/home/nvidia/.local/lib/python3.5/site-packages/pymba/camera.py", line 251, in acquire_frame
    self.AcquisitionStop()
  File "/home/nvidia/.local/lib/python3.5/site-packages/pymba/vimba_object.py", line 34, in <lambda>
    return lambda: self.run_feature_command(item)
  File "/home/nvidia/.local/lib/python3.5/site-packages/pymba/vimba_object.py", line 122, in run_feature_command
    raise VimbaException(error)
pymba.vimba_exception.VimbaException: Timeout during wait.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "capture.py", line 68, in <module>
    capture_frame(camera)
  File "capture.py", line 31, in capture_frame
    camera.arm('SingleFrame')
  File "/home/nvidia/.local/lib/python3.5/site-packages/pymba/camera.py", line 215, in arm
    self.AcquisitionMode = mode
  File "/home/nvidia/.local/lib/python3.5/site-packages/pymba/vimba_object.py", line 49, in __setattr__
    self.feature(item).value = value
  File "/home/nvidia/.local/lib/python3.5/site-packages/pymba/feature.py", line 40, in value
    self._access_func('set', self.info.featureDataType)(value)
  File "/home/nvidia/.local/lib/python3.5/site-packages/pymba/feature.py", line 159, in _set_enum
    raise VimbaException(error)
pymba.vimba_exception.VimbaException: Timeout during wait.
```
The program will frequently capture just one frame, wait until timeout, and through an exception, even if I trigger a frame on the camera's GPIO multiple times.  On Windows, the camera can handle multiple captures sequentially pretty much indefinitely, and never hangs or throws an exception.  Do you have any thoughts on what could be causing this?

morefigs, 2019-05-01:
I've never actually tested outside of Windows, but others that have might be able to help.

NatePolizogo, 2020-11-11:
I have the same problem, I want someone to help too.

morefigs, 2020-11-11:
At this stage I'd recommend using the official VimbaPython support.