morefigs/pymba #95: Pymba cannot find relevant files

lonedevwolf, 2019-06-07:
after i installed pymba and allied vision SDK on a linux environment when i execute

from pymba import Vimba, PYMBA_VERSION

i get the following error 

`Python 3.5.3 (default, Sep 27 2018, 17:25:39)
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pymba import Vimba, PYMBA_VERSION
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.5/site-packages/pymba/__init__.py", line 1, in <module>
    from .vimba import Vimba, VimbaException
  File "/home/pi/.local/lib/python3.5/site-packages/pymba/vimba.py", line 5, in <module>
    from .system import System
  File "/home/pi/.local/lib/python3.5/site-packages/pymba/system.py", line 1, in <module>
    from .vimba_object import VimbaObject
  File "/home/pi/.local/lib/python3.5/site-packages/pymba/vimba_object.py", line 5, in <module>
    from .feature import Feature, _FEATURE_DATA_COMMAND
  File "/home/pi/.local/lib/python3.5/site-packages/pymba/feature.py", line 5, in <module>
    from . import vimba_c
  File "/home/pi/.local/lib/python3.5/site-packages/pymba/vimba_c.py", line 76, in <module>
    vimbaC_path = find_so('arm_32bit', 'GENICAM_GENTL32_PATH')
  File "/home/pi/.local/lib/python3.5/site-packages/pymba/vimba_c.py", line 61, in find_so
    raise OSError('No libVimbaC.so found')
OSError: No libVimbaC.so found
`

maybe the environment path is set incorrectly by allied vision ?
the current path is
`GENICAM_GENTL32_PATH=:/home/pi/Vimba_2_1/VimbaGigETL/CTI/arm_32bit`

Anyone got an idea what went wrong ?

sudheerExperiments, 2019-06-27:
I have the same issue. Did you manage to fix it? @lonedevwolf

sudheerExperiments, 2019-06-27:
> after i installed pymba and allied vision SDK on a linux environment when i execute
> 
> from pymba import Vimba, PYMBA_VERSION
> 
> i get the following error
> 
> `Python 3.5.3 (default, Sep 27 2018, 17:25:39)
> [GCC 6.3.0 20170516] on linux
> Type "help", "copyright", "credits" or "license" for more information.
> 
> > > > from pymba import Vimba, PYMBA_VERSION
> > > > Traceback (most recent call last):
> > > > File "", line 1, in 
> > > > File "/home/pi/.local/lib/python3.5/site-packages/pymba/**init**.py", line 1, in 
> > > > from .vimba import Vimba, VimbaException
> > > > File "/home/pi/.local/lib/python3.5/site-packages/pymba/vimba.py", line 5, in 
> > > > from .system import System
> > > > File "/home/pi/.local/lib/python3.5/site-packages/pymba/system.py", line 1, in 
> > > > from .vimba_object import VimbaObject
> > > > File "/home/pi/.local/lib/python3.5/site-packages/pymba/vimba_object.py", line 5, in 
> > > > from .feature import Feature, _FEATURE_DATA_COMMAND
> > > > File "/home/pi/.local/lib/python3.5/site-packages/pymba/feature.py", line 5, in 
> > > > from . import vimba_c
> > > > File "/home/pi/.local/lib/python3.5/site-packages/pymba/vimba_c.py", line 76, in 
> > > > vimbaC_path = find_so('arm_32bit', 'GENICAM_GENTL32_PATH')
> > > > File "/home/pi/.local/lib/python3.5/site-packages/pymba/vimba_c.py", line 61, in find_so
> > > > raise OSError('No libVimbaC.so found')
> > > > OSError: No libVimbaC.so found
> > > > `
> 
> maybe the environment path is set incorrectly by allied vision ?
> the current path is
> `GENICAM_GENTL32_PATH=:/home/pi/Vimba_2_1/VimbaGigETL/CTI/arm_32bit`
> 
> Anyone got an idea what went wrong ?

Addressed the issue in the below pull request. 

https://github.com/morefigs/pymba/pull/96

It should solve your problem.

morefigs, 2019-06-28:
@sudheerExperiments that fix is for Windows, so won't solve this problem.

sudheerExperiments, 2019-06-28:
@morefigs, sorry didn't notice that the error was related to linux.

lonedevwolf, 2019-07-01:
The issue was that i did not have the right version of vimba installed.

I installed Vimba ARMv7 32 bit and set the GENICAM paths manually since there was some problem with the installer not setting them correctly. Now it is working correctly.

closed 2019-07-01 by lonedevwolf

tmid1, 2024-05-22:
> The issue was that i did not have the right version of vimba installed.
> 
> I installed Vimba ARMv7 32 bit and set the GENICAM paths manually since there was some problem with the installer not setting them correctly. Now it is working correctly.

How did you do this? I'm stuck with the same issue on Raspberry Pi. I tried to follow the command they give in the troubleshooting pdf:

export GENICAM_GENTL64_PATH=$GENICAM_GENTL64_PATH:"/PATH_TO_VIMBAFOLDER/VimbaGigETL/CTI/x86_64bit/"

But I'm not sure it worked.

morefigs, 2024-05-22:
@tmid1 Allied Vision now officially support Python with their cameras, have you tried their software? https://github.com/alliedvision/VmbPy