morefigs/pymba #97: pymba crashes with Vimba 3.0 sdk

sudheerExperiments, 2019-07-09:
Hello,

I recently updated my Vimba SDK . Pymba was working fine with 2.9 but it is crashing with 3.0.

The crash occurs in the below two lines.

```
camera.disarm()
camera.close()
```

Any help would be appreciated. Thanks!

morefigs, 2019-07-10:
Does this example work for you?

sudheerExperiments, 2019-07-10:
It works but when I press ctrl +c to terminate my program, the window just hangs and crashes. This problem wasn't there when I used vimba 2.9.

morefigs, 2019-07-10:
I generally wouldn't expect a program to end nicely upon CTRL+C unless perhaps you're catching the command within Python. Could you post a minimal example that can recreate the bug.

sudheerExperiments, 2019-07-12:
Hello @morefigs,

Here's a sample code of what I am working. I have three cameras in my setup. What I am basically doing in this example is, I am running the three cams in the infinete loop and displaying the grabbed images.

```python
from pymba import Vimba
from pymba import VimbaException
import cv2
import sys

def init_cameras():
        # vimba object
        vimba = Vimba()
        # Start vimba system
        vimba.startup()
        vmFactory = vimba.camera_ids()
        # Get connected cameras
        cams = [vimba.camera(id) for id in vmFactory]
        if len(cams) == 0:
            raise OSError("No camera present.")
        for idx, device in enumerate(vmFactory):
            print("Device {} ID: {}".format(idx, device))
        # Open cameras and print information
        for idx, cam in enumerate(cams):
            try:
                cam.open()
                cam.arm('SingleFrame')
            except VimbaException as e:
                if e.error_code == VimbaException.ERR_TIMEOUT:
                    print(e)
                    cam.disarm()
                    cam.arm('SingleFrame')
                elif e.error_code == VimbaException.ERR_DEVICE_NOT_OPENED:
                    print(e)
                    cam.open()
                    cam.arm('SingleFrame')
        return cams

def main():
    PIXEL_FORMATS_CONVERSIONS = { 'BayerRG8': cv2.COLOR_BAYER_RG2RGB}
    cams = init_cameras()
    flag = True
    while flag:
        try:
            for idx, cam in enumerate(cams):
                frame = cam.acquire_frame()
                img = frame.buffer_data_numpy()
                # convert color space if desired
                img = cv2.cvtColor(img, PIXEL_FORMATS_CONVERSIONS[frame.pixel_format])
                cv2.imshow("Cam {} - {}".format(idx, ["cam{}". format(idx)]), img)
                c = cv2.waitKey(5)
                if c == ord('q'):
                    flag = False
        except KeyboardInterrupt:
            sys.exit(1)
    for cam in cams:
        cam.disarm()
        cam.close()
    cv2.destroyAllWindows()

if __name__ == '__main__':
    main()
```

**Sample output:**

Device 0 ID: DEV_000F31008103
Device 1 ID: DEV_000F31008105
Device 2 ID: DEV_000F31008104

cmd prompt just hangs when I hit '**ctrl + c**' or '**q**'.

You should be able to replicate the issue with this example. Let me know if you have any questions.

sudheerExperiments, 2019-07-18:
@morefigs Any update on this issue?

morefigs, 2019-08-19:
Sorry, haven't had time to debug this and likely won't soon. I can't see any obvious reason this wouldn't work, however.

SunnyAVT, 2019-08-22:
@sudheerExperiments, I copy and run your code in my test PC(win10 Pro 64bit, Python3.6 and Vimba SDK 3.0, I hit 'q', it has not any problem, not crash. The test result is as attached picture.
Maybe you can give me more info for how to re-produce the issue.
[image]
BTW, the previous Vimba SDK is V2.1.3, not V2.9.

kthumma2, 2023-06-02:
Hello , 
Can anyone help me out how to set up the path for Vimba 6.0 on my windows ? I Have installed Vimba SDK but I am unable to set path for the bin. i am trying to interact this Vimba with my python code i am unable to do it. 
Any suggestions?