morefigs/pymba #98: camera.arm("SingleFrame") invalid with current access mode.

bsounak, 2019-08-06:
from pymba import Vimba

    with Vimba() as vimba:
        vimba.startup()

        print("vimba version - {}".format(vimba.version()))
        cam = vimba.camera("10.10.20.12")
        cam.open()

        feature = cam.feature("PixelFormat")
        print("PixelFormat - {}".format(feature.value))

        cam.arm("SingleFrame")
        frame = cam.acquire_frame()

    cam.close()

The output is

(env1) C:\Users\foo\Desktop>python -m allied
vimba version - 1.7.0
PixelFormat - Mono12
Traceback (most recent call last):
  File "c:\users\foo\appdata\local\programs\python\python36\Lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\foo\appdata\local\programs\python\python36\Lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\foo\Desktop\allied.py", line 15, in <module>
    cam.arm("SingleFrame")
  File "C:\Users\foo\Envs\env1\lib\site-packages\pymba\camera.py", line 224, in arm
    self.start_capture()
  File "C:\Users\sbhatt\Envs\env1\lib\site-packages\pymba\camera.py", line 164, in start_capture
    raise VimbaException(error)
pymba.vimba_exception.VimbaException: Operation is invalid with the current access mode.

morefigs, 2019-08-07:
Is anything else using the camera at the same time?

You should remove `vimba.startup()` as that's done already within `with Vimba() as vimba:`.

bsounak, 2019-08-07:
No. I have restarted both the computer and the switch and it still gives this error. 

How do I check the current access mode? This is the list of access modes I could find from the camera object.

```
print("VMB_ACCESS_MODE_FULL - {}".format(cam.VMB_ACCESS_MODE_FULL))
print("VMB_ACCESS_MODE_CONFIG - {}".format(cam.VMB_ACCESS_MODE_CONFIG))
print("VMB_ACCESS_MODE_LITE - {}".format(cam.VMB_ACCESS_MODE_LITE))
print("VMB_ACCESS_MODE_NONE - {}".format(cam.VMB_ACCESS_MODE_NONE))
print("VMB_ACCESS_MODE_READ - {}".format(cam.VMB_ACCESS_MODE_READ))	
```

VMB_ACCESS_MODE_FULL - 1
VMB_ACCESS_MODE_CONFIG - 4
VMB_ACCESS_MODE_LITE - 8
VMB_ACCESS_MODE_NONE - 0
VMB_ACCESS_MODE_READ - 2

Strangely I have another manta camera of the same model, for which the camera.arm("SingleFrame") works and it outputs the same access mode status values.

bsounak, 2019-08-07:
It was the switch. I was using a ZyXEL ES1100-8P. Changing it to tp-link TL-SG108PE solved the issue. I have no idea why.

bsounak, 2019-08-07:
I changed the switch and it seems to work now. Not sure why though, I was able to use vimba viewer and capture images with the old switch.

closed 2019-08-10 by bsounak