morefigs/pymba #100: Fail to get correct data with >8 bit depth color pixel format

SunnyAVT, 2019-08-16:
I upload my code to show the image from color camera.
frame.buffer_data_numpy cannot return correct data if I use more than 8 bit depth pixel format, such as  "BayerRG12", "RGB8Packed" and "RGB8Packed".
The code can show correct image at "BayerRG8" format.

from __future__ import absolute_import, print_function, division
from pymba import *
import numpy as np
import cv2
import time

Vimba().startup()
camera = Vimba().camera(0)
camera.open()
print("Camera List ID is %s", Vimba.camera_ids())

pixel_format = camera.feature("PixelFormat")
pixel_format.value = "BayerRG8"
# pixel_format.value = "BayerRG12"
# Note: Fail to work at "BayerRG12" "RGB8Packed" "RGB8Packed" format

camera.arm('SingleFrame')

# capture a single frame, more than once if desired
for i in range(1):
    frame = camera.acquire_frame()
    print('frame {}'.format(frame.data.frameID))
    frame_size_camera = len(frame.buffer_data())
    frame_pixel_format = frame.pixel_format
    print("image size %d, pixel_format: %s" % (frame_size_camera, frame_pixel_format))

    # get a copy of the frame data
    img = frame.buffer_data_numpy()
    print("image buffer: size is ", img.shape)

    cv2.imshow("testBayer", img)
    k = cv2.waitKey(50)

    colorImg = cv2.cvtColor(img, cv2.COLOR_BAYER_RG2RGB )
    cv2.imshow("testRGB", colorImg)
    k = cv2.waitKey(0)

camera.disarm()
camera.close()
Vimba.shutdown()

morefigs, 2019-08-18:
Yes, greater than 8-bit images are not supported yet. If you manage to get it worked please add a pull request.

honkomonk, 2019-08-21:
I just opened a pull request that would add support for most pixel formats.  -> #102

SunnyAVT, 2019-08-23:
I saw it, it is great! Thanks.