morefigs/pymba #103: Using pymba with opencv trackers

smsalaken, 2019-08-23:
Hi, 
I have a use for pymba to track object on opencv. This needs to work similar to cv.VideoCapture() and allow the user to select an object using selectROI() method. I couldn't figure out how to get it done using pymba. 
Could you please give me some hint or example code to start with?
Thanks

SunnyAVT, 2019-08-23:
You can refer to attached code for how to access the camera, get image and set ROI. More samples can refer to https://github.com/morefigs/pymba/tree/master/examples/camera

from pymba import Vimba
from pymba import VimbaException
import cv2
import sys

def Camera_Set_ROI(camera, W: int, H: int) -> None:
    image_Height = camera.feature("Height")
    image_Width = camera.feature("Width")
    image_Height.value = H
    image_Width.value = W

def init_cameras():
        # vimba object
        vimba = Vimba()
        # Start vimba system
        vimba.startup()
        vmFactory = vimba.camera_ids()
        # Get connected cameras
        cams = [vimba.camera(id) for id in vmFactory]
        if len(cams) == 0:
            raise OSError("No camera present.")
        for idx, device in enumerate(vmFactory):
            print("Device {} ID: {}".format(idx, device))
        # Open cameras and print information
        for idx, cam in enumerate(cams):
            try:
                cam.open()
                Camera_Set_ROI(cam, 640, 480)
                cam.arm('SingleFrame')
            except VimbaException as e:
                if e.error_code == VimbaException.ERR_TIMEOUT:
                    print(e)
                    cam.disarm()
                    cam.arm('SingleFrame')
                elif e.error_code == VimbaException.ERR_DEVICE_NOT_OPENED:
                    print(e)
                    cam.open()
                    cam.arm('SingleFrame')
        return cams

def main():
    PIXEL_FORMATS_CONVERSIONS = {'BayerRG8': cv2.COLOR_BAYER_RG2RGB}
    cams = init_cameras()
    flag = True
    while flag:
        try:
            for idx, cam in enumerate(cams):
                frame = cam.acquire_frame()
                img = frame.buffer_data_numpy()
                # convert color space if desired
                if(frame.pixel_format == 'BayerRG8'):
                    img = cv2.cvtColor(img, PIXEL_FORMATS_CONVERSIONS[frame.pixel_format])
                img = cv2.resize(img, (400, 300))
                cv2.imshow("Cam {} - {}".format(idx, ["cam{}". format(idx)]), img)
                c = cv2.waitKey(5)
                if c == ord('q'):
                    flag = False
        except KeyboardInterrupt:
            sys.exit(1)
    for cam in cams:
        cam.disarm()
        cam.close()
    cv2.destroyAllWindows()
    Vimba().shutdown()

if __name__ == '__main__':
    main()

SunnyAVT, 2019-08-23:
cv.VideoCapture() cannot be used to access Pymba camera. You can refer to attached code for how to access the camera, get image and set ROI (Camera_Set_ROI function as below). More samples can refer to https://github.com/morefigs/pymba/tree/master/examples/camera

from pymba import Vimba
from pymba import VimbaException
import cv2
import sys

def Camera_Set_ROI(camera, W: int, H: int) -> None:
    image_Height = camera.feature("Height")
    image_Width = camera.feature("Width")
    image_Height.value = H
    image_Width.value = W

def init_cameras():
        # vimba object
        vimba = Vimba()
        # Start vimba system
        vimba.startup()
        vmFactory = vimba.camera_ids()
        # Get connected cameras
        cams = [vimba.camera(id) for id in vmFactory]
        if len(cams) == 0:
            raise OSError("No camera present.")
        for idx, device in enumerate(vmFactory):
            print("Device {} ID: {}".format(idx, device))
        # Open cameras and print information
        for idx, cam in enumerate(cams):
            try:
                cam.open()
                Camera_Set_ROI(cam, 640, 480)
                cam.arm('SingleFrame')
            except VimbaException as e:
                if e.error_code == VimbaException.ERR_TIMEOUT:
                    print(e)
                    cam.disarm()
                    cam.arm('SingleFrame')
                elif e.error_code == VimbaException.ERR_DEVICE_NOT_OPENED:
                    print(e)
                    cam.open()
                    cam.arm('SingleFrame')
        return cams

def main():
    PIXEL_FORMATS_CONVERSIONS = {'BayerRG8': cv2.COLOR_BAYER_RG2RGB}
    cams = init_cameras()
    flag = True
    while flag:
        try:
            for idx, cam in enumerate(cams):
                frame = cam.acquire_frame()
                img = frame.buffer_data_numpy()
                # convert color space if desired
                if(frame.pixel_format == 'BayerRG8'):
                    img = cv2.cvtColor(img, PIXEL_FORMATS_CONVERSIONS[frame.pixel_format])
                img = cv2.resize(img, (400, 300))
                cv2.imshow("Cam {} - {}".format(idx, ["cam{}". format(idx)]), img)
                c = cv2.waitKey(5)
                if c == ord('q'):
                    flag = False
        except KeyboardInterrupt:
            sys.exit(1)
    for cam in cams:
        cam.disarm()
        cam.close()
    cv2.destroyAllWindows()
    Vimba().shutdown()

if __name__ == '__main__':
    main()

kumbhat941, 2024-09-11:
I am trying to set roi using the given program but with singleframe the image is completely dark, in my view may be due to lack of time to adjustment. So I tried using continuous mode for it but it does not work. Can you suggest me a better approach ?