morefigs/pymba #104: Can't open camera after setting multicast

CDPAlolita, 2019-09-04:
**Dear pymba team:**
Thank you for making such brilliant package.
It's convenient and easy learning.
I used **pymba** for a while, it worked well.
I could acquire the video stream with my python program smoothly.

But my team decided to share the screen with different users.
So, I use **Vimba** to set the multicast.
After then, I ran the example  **"opencv_acquire_streaming_images.py"** from this site.
And it stuck at the camera opening step.
Saying **"pymba.vimba_exception.VimbaException: Operation is invalid with the current access mode."**

Below is the part of the code: 
```
from time import sleep
from pymba import Vimba
from examples.camera.display_frame import display_frame

if __name__ == '__main__':

    with Vimba() as vimba:
        camera = vimba.camera(0)
        camera.open() #--------The error line--------#
```
I am not sure if this is because of my **Vimba** setting incorrectly, or **pymba** doesn't support multicast?

I didn't found any answer at internet or this site.
So, sorry if the answer is already existed.
And also sorry for my poor English.

morefigs, 2019-09-05:
I've never tried so can't say for sure, however:
- which OS?
- you may need to call `open` with a different access mode - the default is `VimbaObject.VMB_ACCESS_MODE_FULL`, which is possibly the issue here.

CDPAlolita, 2019-09-05:
Dear morefigs:

* which OS?

    **Windows 10 Professional, 64-bit**

* do you already have a "master" instance of vimba viewer connected to the camera?

    **I am not sure what does "master" instance mean.
    I only know that I can use vimba viewer to see the video stream, but my python program can't acquilre that**

* you may need to call open with a different access mode - the default is VimbaObject.VMB_ACCESS_MODE_FULL, which is possibly the issue here.

    **I tried that, and these is my code and result:**

    ```
    from time import sleep
    from pymba import Vimba, vimba_object
    from examples.camera.display_frame import display_frame

    if __name__ == '__main__':

        with Vimba() as vimba:
            VO = vimba_object.VimbaObject(vimba) #Add the VimbaObject
            camera = vimba.camera(0)
            camera.open(camera_access_mode = VO.VMB_ACCESS_MODE_READ) #Change the access mode

I tried **VMB_ACCESS_MODE_(FULL, READ, CONFIG, LITE)**.
They gave me the same error message --- **"VimbaException: Operation is invalid with the current access mode."**

And the **VMB_ACCESS_MODE_NONE** gave me the error like ---**"VimbaException: API feature is not implemented."**
I guess these are all the modes I can try.

SunnyAVT, 2019-09-05:
The access-mode of the camera provides information in which mode the camera "can" be opened. So before opening the camera, the full access mode  will be available. After opening the camera from the current application instance, the camera can not opened anymore at full access mode because it is already opened by the instance of the instance which will result in "no access".

SunnyAVT, 2019-09-05:
You can open camera in pymba or open camera in other application, but cannot be both. :(-

CDPAlolita, 2019-09-05:
**Dear SunnyAVT:**
So, there's no way to turn off the the redundant acquirement.
And make my python program like **Vimba viewer**, which can see other computer's multicast data but not to access the camera...

Thank you for replying.
I will figure other way to reach my target.