morefigs/pymba #115: Unable to install pympa on raspberry  pi

zjw-asu, 2019-11-08:
Hi there,
I have tried to install pymba using pip install pymba on raspberry pi command line but failed. And it indicates a syntax error in setup.py. 
Do you support raspberry pi with Raspbian (a linux system for raspberry pi) ? If so could you let me know how to install pymba correctly on raspberry pi? Thanks !

morefigs, 2019-11-10:
Hi, does it work for you on PC? I have yet to try RPi.

zjw-asu, 2019-11-10:
Hi, I have tried to install using anaconda on my windows system. So far it works as I can read the vimba version and read the camera ID. I have yet tried to get images as I had problems installing OpenCV.But I believe it will work on windows too.

morefigs, 2019-11-10:
Could you please post the error traceback?

zjw-asu, 2019-11-10:
Please see traceback below:
pi@raspberrypi:~ $ pip install pymba
Collecting pymba
  Using cached https://files.pythonhosted.org/packages/04/78/d29c37c6419467d60a018a8da98c423f12b64491b4a5dc01b275b3f44d36/pymba-0.3.6.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-MCC3Dc/pymba/setup.py", line 2, in <module>
        from pymba import __version__
      File "pymba/__init__.py", line 1, in <module>
        from .vimba import Vimba, VimbaException
      File "pymba/vimba.py", line 17
        def version() -> str:
                      ^
    SyntaxError: invalid syntax
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-MCC3Dc/pymba/

morefigs, 2019-11-11:
It looks like it's getting thrown by the type hint. Are you using Python 2? If so I'd strongly recommend using Python 3 instead. Alternatively, consider using the `python2` branch of Pymba (although this is not well supported).

zjw-asu, 2019-11-11:
Hi , 
Indeed It's because of version problem. I have cloned python2 branch and installed the pymba successfully. However, it shows another error when I try to run :
 from pymba import *
    
    with Vimba() as vimba:
        print(vimba.getVersion())
The error is below 
Traceback (most recent call last):
  File "test_installation.py", line 3, in <module>
    from pymba import Vimba
  File "/usr/local/lib/python2.7/dist-packages/pymba-0.1-py2.7.egg/pymba/__init__.py", line 3, in <module>
    from .vimba import Vimba
  File "/usr/local/lib/python2.7/dist-packages/pymba-0.1-py2.7.egg/pymba/vimba.py", line 4, in <module>
    from .vimba_dll import VimbaDLL
  File "/usr/local/lib/python2.7/dist-packages/pymba-0.1-py2.7.egg/pymba/vimba_dll.py", line 60, in <module>
    "GENICAM_GENTL32_PATH"), "you need your GENICAM_GENTL32_PATH environment set.  Make sure you have Vimba installed, and you have loaded the /etc/profile.d/ scripts"
AssertionError: you need your GENICAM_GENTL32_PATH environment set.  Make sure you have Vimba installed, and you have loaded the /etc/profile.d/ scripts

I have installed pymba on raspberry pi before and I manage to obtain images. So I guess current code may not work for raspberry pi. But, Much appreciation to your generous help!!

morefigs, 2019-11-11:
That error generally means you don't have Vimba installed, or the installation cannot be found. Could be a bug. Do you definitely have Vimba installed? I'm not so familiar with installing on Linux/RPi.

SunnyAVT, 2019-11-11:
You have to install Vimba ARM SDK before running Pymba.
You can get Vimba for arm sdk from https://www.alliedvision.com/en/products/software.html
As an example, Downloads 
for ARMv7 32-bit:
https://www.alliedvision.com/en/products/software.html#agb-modal-content-5496
for ARMv8 64-bit: 
https://www.alliedvision.com/en/products/software.html#agb-modal-content-5496
Some instruction refer to https://cdn.alliedvision.com/fileadmin/content/documents/products/software/software/Vimba/appnote/Vimba_installation_under_Linux.pdf
After Vimba SDK was installed correctly, then reboot, next to check the "GENICAM_GENTL32_PATH" setting, and run VimbaViewer coming with SDK for the correction.
Then you can run Pymba again.

zjw-asu, 2019-11-12:
Hi, 
I managed to run code on the raspberry pi with python 2 now!!! Much thanks!

closed 2019-11-12 by zjw-asu