morefigs/pymba #116: Operation is invalid with the current access mode when doing start_capture()

GGGMatt89, 2019-11-12:
Hi all, 
I have a Prosilica GT1660 and I am trying to acquire a single image with pymba. I am able to read and set different features like PixelFormat,  GVSPPacketSize and AcquisitionMode, but when trying to run a simple acquisition, it is aborted with the error reported in the title. I tried with both my own code and the example code in opencv_acquire_image.py, same result. 

Thank you in advance for your help!

morefigs, 2019-11-12:
I've never tested those cameras. Do you have the camera open anywhere else at the same time?

GGGMatt89, 2019-11-13:
No, but I am working on a virtual machine, I don’t know if this can affect in some way

elliot-proton, 2019-11-18:
Are you able to connect in _full access mode_ via Vimba Viewer?

GGGMatt89, 2019-11-19:
Yes, and I can launch an acquisition from there. Then the acquisition ends with frame errors, but I can start at least. 

Mattia Fontana, PhD

TORET Devices srl
Mail: mattia.fontana@toretdevices.com
Web: www.toretdevices.com 
Tel: +39 011 19907880
Sede operativa: Lungo Dora Voghera 36A, 10153 TORINO