morefigs/pymba #121: No transport layers were found

electronsandstuff, 2020-02-15:
I'm trying to use pymba on Windows w/ Python 3.7.  I followed the installation instructions, but receive the following error when trying to setup the library.
[image]
I am able to see the camera I want to get data from in both Vimba Viewer and with the python library harvesters.  I have the GigE drivers installed and have also tried removing all other transport drivers besides the one I need following the advice of a previous issue.

morefigs, 2020-02-16:
Does it work in Vimba Viewer? Have you rebooted since installing the driver?

electronsandstuff, 2020-02-17:
Yeah, it works perfectly in Vimba Viewer and when I use another python library `harvester`.  But, good point.  I haven't tried rebooting yet, I can check that out this week.