morefigs/pymba #122: unable to change OffsetX or OffsetY

skywo1f, 2020-03-23:
In the vimba viewer, there is an option to change the offset of the pixel-screen where I would like to pull data from. These are called OffsetX and OffsetY. I tried to change them via:

        # read a feature value
        featureOY = camera.feature('OffsetY')
        value = featureOY.value
        # set the feature value
        featureOY.value = 1                      #404

But I get:
/mnt/xavier1ssd/pymba-cam/examples/camera$ python3 camera_color_video.py 
Traceback (most recent call last):
  File "camera_color_video.py", line 63, in <module>
    featureOY.value = 1                      #404
  File "/usr/local/lib/python3.6/dist-packages/pymba/feature.py", line 40, in value
    self._access_func('set', self.info.featureDataType)(value)
  File "/usr/local/lib/python3.6/dist-packages/pymba/feature.py", line 125, in _set_int
    raise VimbaException(error)
pymba.vimba_exception.VimbaException: The value was not valid; either out of bounds or not an increment of the minimum.

setting them to 0 works fine, but I just see the upper corner of the vision range instead of the whole thing.

jakove, 2020-03-25:
Before you can set the offset yu have to change the image size.

For example:
If you want to set "OffsetY" to 200 you have to decrease "Height" by 200.

closed 2020-03-26 by morefigs

skywo1f, 2020-03-27:
That worked. I was changing the height and width, but I was doing it after the offset. When I changed it to before, everything worked.