morefigs/pymba #124: get timestamp of a frame

NejcKozjekKnaufInsulation, 2020-04-29:
I am trying to get a timestamp of a frame, but can't find how. I am running the camera in continuous mode.

Currently I am doing it this way and I don't know if the timestamp I am getting is equal to the real timestamp.
```
def sv_frame(self, frame):
    img = frame.buffer_data_numpy()
    timestamp = time.time()
```
I noticed in Vimba there's a command *GevTimestampControlLatch* and a variable *GevTimestampValue* which should control the appending timestamp to a frame and storing it into a variable. However I cannot run the aforementioned command and the variable is always 0.

Looking at a source code I have found one reference to timestamp and I was able to acquire it via frame.data.timestamp. Per Vimba documentation I divided the number with the *GevTimestampTickFrequency* of the oscillator, but I receive a float number that doesn't resemble the timestamp. However the difference in consequent frames do reflect the exposure time of the camera (in my case 3-4 ms).
```
def sv_frame(self, frame):
    img = frame.buffer_data_numpy()
    timestamp = frame.data.timestamp / self.tick_frequency
```
Example output:
1) Timestamp from time.time(): 1588177322.3672035
2) Timestamp from second approach: 19806.7125729

"1" yields 18:22:2.367203
"2" yields (considering time in seconds): 5:30:6.7125728

How can I get a sensible timestamp from the camera?

Dradeliomecus, 2020-08-24:
I'm also interested in this feature,
Is there any plan to implement this in pymba in the near future?

NejcKozjekKnaufInsulation, 2020-08-25:
I have enquired with Allied Vision regarding the timestamp acquisition and their response was that the cameras are not capable of outputting the timestamp. 

Here is code snippet on how I resolved the camera's timestamps in the callback function for arming the camera:
	
```
                # timestamp from camera
		timestamp = frame.data.timestamp / self.timestamp_frequency
		
		# this is an example how to convert timestamp from camera to a POSIX format
		if self.start_capture:
			# mark the begining of camera capture
			self.timestamp_reference = time.time()
			# get the difference between the two entities
			self.timestamp_mapping = self.timestamp_reference - timestamp
			# mark that camera capture has started
			self.start_capture = False

		# map camera timestamp to POSIX timestamp
		timestamp = self.timestamp_mapping + timestamp
```

Hope this helps.

Cheers

Dradeliomecus, 2020-08-25:
In the Vimba CPP Manual (4.1), section 5.11.22, it is described that there is a `Frame.GetTimeStamp()` function returning a `VmbUint64_t&`.

Isn't it possible for `pymba` to have access to this?

NejcKozjekKnaufInsulation, 2020-08-25:
If the function you mention is returning a 64 bit unsigned integer, then it cannot contain the millisecond information. I did not look into the function, but I assume it returns the internal counter, which can be obtained with pymba with frame.data.timestamp. By incorporating the tick frequency, you can transform the counter into a time based scale.