morefigs/pymba #130: Streaming with 'Continuous' mode disconnect camera in few seconds

oao112358, 2020-09-16:
Hi, I'm trying to to use pymba with Manta G-419C(Gige).

For instance, when I do `opencv_acquire_streaming_images.py` in examples, and I use Vimba Viewer to confirm that camera is locked by python or not. I can get images in first few seconds(about 2 or 3 seconds), and the camera is actually locked by python. But after that, I fount that the camera is unlocked by Vimba Viewer, and my python cannot get other image anymore. 

Then I will get `Operation is invalid with the current access mode` on `camera.stop_frame_acquisition()` or `camera.disarm()`. If I comment out these codes, python will close without any exception or error. So I guess that `Operation is invalid with the current access mode` is caused by python didn't get camera's control. 

However, I can get streaming by Vimba Viewer and Matlab uninterrupted without any error.
I really tried to search a solution on internet or other issue but I can't find any similar problem.

Thank you for your help!

mvalvaa, 2020-09-21:
Hi, I'am using an Alvium 1800U - 1240m

I get the same error when trying to stream images in 'Continuous' mode. If I arm the camera in 'SingleFrame' I can use cv2.imshow in loop to view live images from the camera but i get lower fps then what the camera can do (i get 5, max is 29).

Actually in 'Continuous' mode in Vimba Viewer a I also get 5 fps. And I think it's not a exposure time realated problem.
When I try 'Continuous' mode in python I get `Operation is invalid with the current access mode`.

Maybe it's not the same error idk. I'm new using python and pymba.

Just need some help.

Thanks.

rk-exxec, 2020-10-12:
VimbaViewer will take control of the camera if you press play/stop once. You have to close vimba viewer and reopen the cam in python.
If you look at the official VimbaPython, it basically forces you to reopen the API and camera connection every time you want to send a command. I guess thats whats happening with MATLAB, too. It forcibly takes control of the cam.

I had this issue and solved it by calling the AcquisitionStop command feature directly.