morefigs/pymba #133: python 2 camera setting upload

rwreynol, 2020-11-02:
I do not see a method for uploading a settings file in the python 2 branch. I need to use python 2 so switching to 3 is not an option. Is there a method for uploading an XML settings file that I overlooked in the python2 branch.

morefigs, 2020-11-02:
No, there doesn't appear to be. Happy to accept PRs though :)

rwreynol, 2020-11-03:
I would appreciate the ability to load an XML file greatly. I gave it a
shot yesterday with no success. I tried using your Python3 code as a
template.

Randall

rwreynol, 2020-11-18:
Hello,

I don't know if you have added the settings upload for python2, but I have
added a function to the vimba_camera object to upload XML settings files if
you are interested. I can send you the changes.

Randall

morefigs, 2020-11-18:
Sure, thanks, feel free to add a pull request.